zephyrproject-rtos/zephyr #64940: twister: test: pytest -s scripts/tests/twister/test_runner.py fails

yperess, 2023-11-07:
**Describe the bug**
pytest -s scripts/tests/twister/test_runner.py fails

**To Reproduce**
Steps to reproduce the behavior:

1. run below
    pytest -s scripts/tests/twister/test_runner.py

**Expected behavior**
test pass

**Impact**
CI

**Logs and console output**
```
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_log_info[basic] - TypeError: ProjectBuilder.log_info() missing 1 required positional argument: 'inline_logs'
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_log_info[inline logs] - TypeError: ProjectBuilder.log_info() missing 1 required positional argument: 'inline_logs'
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_log_info[inline logs+read_exception] - TypeError: ProjectBuilder.log_info() missing 1 required positional argument: 'inline_logs'
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[verbose error cmake only] - assert False
 +  where False = all([False])
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[failed] - assert False
 +  where False = all([False])
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[verbose skipped] - assert False
 +  where False = all([False])
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[verbose passed ready run] - assert False
 +  where False = all([False])
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[verbose passed] - assert False
 +  where False = all([False])
FAILED scripts/tests/twister/test_runner.py::test_projectbuilder_report_out[timeout] - assert False
 +  where False = all([False])
```

Git bisect returns the following as the commit that broke things
```
commit e466b7ac266b24fe4e1c27b01fef3a722e18ddd3
Author: Grzegorz Chwierut <grzegorz.chwierut@nordicsemi.no>
Date:   Thu Oct 26 09:08:54 2023 +0200

    twister: pytest: Improve reporting failed pytest scenarios
    
    When pytest scenario fails, then 'handler.log' is printed.
    Changed to print 'twister_harness.log' that is a log from
    pytest execution. That file tells much more when test fails.
    
    Signed-off-by: Grzegorz Chwierut <grzegorz.chwierut@nordicsemi.no>

 .../src/twister_harness/device/hardware_adapter.py |  1 -
 scripts/pylib/twister/twisterlib/harness.py        |  2 ++
 scripts/pylib/twister/twisterlib/reports.py        |  5 ++++-
 scripts/pylib/twister/twisterlib/runner.py         | 26 +++++++++++++++++-----
 .../pytest_integration/test_harness_pytest.py      |  2 ++
 5 files changed, 29 insertions(+), 7 deletions(-)
```

nashif, 2023-11-07:
Something strange is going on...

The PR introducing the commit above did not fail on the tests (https://github.com/zephyrproject-rtos/zephyr/actions/runs/6669873320/job/18128568198) but, I can see that a different number of tests is being run in CI compared when I run on the same commit locally:

in CI:
```
====================== 355 passed, 278 warnings in 9.31s =======================
```

locally:

```
======================= 9 failed, 505 passed, 273 warnings in 17.70s =====================
```

hakehuang, 2023-11-08:
this PR may fix this. #64947 64947

yperess, 2023-11-08:
Seems to be passing now.

closed 2023-11-08 by yperess

LukaszMrugala, 2023-11-09:
> Something strange is going on...
>
> The PR introducing the commit above did not fail on the tests (https://github.com/zephyrproject-rtos/zephyr/actions/runs/6669873320/job/18128568198) but, I can see [...]

I have investigated this issue and it seems that nothing is out of the ordinary. The PR introducing the commit did not fail any test two weeks ago. However, between then and now, about 187 new tests have been introduced to the mainline - (`test_error.py`, `test_hardwaremap.py`, `test_jobserver.py` and new tests in `test_harness_pytest.py`). Those too passed their checks, as those occurred before the introduction of the aforementioned commit.

Only when both changes were live, the tests started failing. Not sure how we can prevent this. Manual Checks re-run before merging? Automatic Checks rerun when attempting merge with abort on fail? Any mainline merge reruns all PR Checks?