zephyrproject-rtos/zephyr #64952: ESP32-S3-DevKitC-1U-N8R2 can not boot to zephyr os

sam131208, 2023-11-08:
**Describe the bug**
 Build the project zephyr\samples\bluetooth\peripheral_hids and flash. Cannot boot to Zephyr OS.
 It looks like it's constantly reset.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. git pull
3. west update
4. cd zephyr\samples\bluetooth\peripheral_hids
5.  west build -b esp32s3_devkitm
6.  west flash
7. See error

**Expected behavior**
 Successfully booted to the Zephyr OS.

**Logs and console output**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x164c
load:0x403b6000,len:0xc68
load:0x403ba000,len:0x30e8
SHA-256 comparison failed:
Calculated: e4ce81ff053684236d4bb74e7cc1c3e90cc1bfe9088459f3d355d6c7f4b209a9
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403b61f4
I (48) boot: ESP-IDF b3cb13f06 2nd stage bootloader
I (48) boot: compile time 16:16:02
I (48) boot: chip revision: 0
I (50) boot.esp32s3: Boot SPI Speed : 40MHz
I (55) boot.esp32s3: SPI Mode       : DIO
I (60) boot.esp32s3: SPI Flash Size : 16MB
I (65) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)
I (116) esp_image: segment 1: paddr=00010044 vaddr=3fc9d160 size=00ee8h (  3816) load
I (125) esp_image: segment 2: paddr=00010f34 vaddr=3fc9e048 size=00478h (  1144) load
I (133) esp_image: segment 3: paddr=000113b4 vaddr=40374000 size=091b0h ( 37296) load
I (152) esp_image: segment 4: paddr=0001a56c vaddr=00000000 size=05acch ( 23244)
I (158) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=06ce8h ( 27880) map
I (166) esp_image: segment 6: paddr=00026d30 vaddr=00000000 size=092c8h ( 37576)
I (175) esp_image: segment 7: paddr=00030000 vaddr=42030000 size=23a6ch (146028) map
I (216) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd0108,len:0x164c
load:0x403b6000,len:0xc68
load:0x403ba000,len:0x30e8
SHA-256 comparison failed:
Calculated: e4ce81ff053684236d4bb74e7cc1c3e90cc1bfe9088459f3d355d6c7f4b209a9
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403b61f4

**Environment (please complete the following information):**
 - OS:  Windows
 - Toolchain: Zephyr SDK 0.16.3
 - The latest version of Zephyr.
 - Board: ESP32-S3-DevKitC-1U-N8R2

sylvioalves, 2023-11-08:
@sam131208 Do you have any modifications in your prj.conf? I just tested and it worked properly (Linux):  
```c
*** Booting Zephyr OS build zephyr-v3.5.0-1144-g4df07224f8dc ***
[00:00:00.312,000] <inf> fs_nvs: 6 Sectors of 4096 bytes
[00:00:00.312,000] <inf> fs_nvs: alloc wra: 0, fd0
[00:00:00.312,000] <inf> fs_nvs: data wra: 0, 1c
[00:00:00.313,000] <inf> esp32_bt_adapter: BT controller compile version [d913766]
[00:00:00.485,000] <inf> bt_hci_core: No ID address. App must call settings_load()
Bluetooth initialized
[00:00:00.487,000] <inf> bt_hci_core: Identity: 68:B6:B3:37:F0:AE (public)
[00:00:00.487,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0016, manufacturer 0x0060
[00:00:00.487,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0x0016
Advertising successfully started
Bluetooth authentication callbacks registered.
```

sam131208, 2023-11-08:
@sylvioalves  No changes have been made to any of the files, including prj.conf.
In fact, on my personal computer, it works fine. 
However, on my work computer, whether it's Windows or Ubuntu, this issue occurs. Even I've reinstalled both operating systems.
It is worth noting that when running this project zephyr\samples\hello_world, it works fine. However, this problem will occur for any project in the zephyr\samples\Bluetooth directory.

sylvioalves, 2023-11-08:
Well, if it works in your personal computer and also in mine, I guess other computers might have something wrong?  
Regarding the bluetooth sample, can you retry modifying the `SOC_PART_NUMBER`to:  
`west build -b esp32s3_devkitm samples/bluetooth/peripheral_hids/ -DCONFIG_SOC_ESP32S3_PICO_N8R2=y`

sam131208, 2023-11-09:
@sylvioalves Compile according to your modifications, and the result is still the same.

I compared the zephyr.map file of the two computers and found that they were the same except for the debug information.
I'm guessing it might be an alignment issue with the debug info?
The following zephyr .zip file contains two files zephyr_ok.map and zephyr_bad.map, which may be helpful for your analysis.

[zephyr.zip](https://github.com/zephyrproject-rtos/zephyr/files/13304951/zephyr.zip)

sylvioalves, 2023-11-09:
Both computers point to the same `zephyr`and `hal_espressif` hashes/revision?

sam131208, 2023-11-09:
Yes, the version is consistent. Before compiling, only the latest version is updated with git pull, and the interval between updates is two hours.

sam131208, 2023-11-15:
@sylvioalves The problem has been solved. The board connects to the USB 3.0 interface and works fine.The guess is that the USB 2.0 port is underpowered.

closed 2023-11-15 by sam131208

bienxanh1901, 2024-04-11:
Hi Everyone,
I also faced in to the same problem with ESP32-S3-DevKitC-1-N16R2.
Just run the hello world sample but cannot flash to the board. The log is below:
`SHA-256 comparison failed:
Calculated: 6b1c323e009a935db2ddeaee7f98520c0389db7dd875a9f0f5db6e6b6ab9838f
Expected: 0000000040540000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x403778e4
0x403778e4: __start at C:/Users/Administrator/Documents/zephyrproject/zephyr/soc/espressif/common/loader.c:218

I (64) boot: ESP Simple boot
I (64) boot: compile time Apr 11 2024 09:05:36
W (64) boot: Unicore bootloader
I (65) spi_flash: detected chip: winbond
I (66) spi_flash: flash io: dio
W (69) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (81) boot: chip revision: v0.2
I (84) boot.esp32s3: Boot SPI Speed : 40MHz
I (88) boot.esp32s3: SPI Mode       : DIO
I (92) boot.esp32s3: SPI Flash Size : 8MB
I (95) boot: Enabling RNG early entropy source...
[esp32s3] [INF] DRAM: lma 0x00000020 vma 0x3fc8d338 len 0x186c   (6252)
[esp32s3] [INF] IRAM: lma 0x00001894 vma 0x40374000 len 0x9318   (37656)
0x40374000: _WindowOverflow4 at C:/Users/Administrator/Documents/zephyrproject/zephyr/arch/xtensa/core/window_vectors.S:56

[esp32s3] [INF] padd: lma 0x0000abb8 vma 0x00000000 len 0x5440   (21568)
[esp32s3] [INF] IMAP: lma 0x00010000 vma 0x42000000 len 0x38b0   (14512)
0x42000000: _stext at ??:?

[esp32s3] [INF] padd: lma 0x000138b8 vma 0x00000000 len 0xc740   (51008)
[esp32s3] [INF] DMAP: lma 0x00020000 vma 0x3c010000 len 0x1140   (4416)
[esp32s3] [INF] Image with 6 segments
[esp32s3] [INF] DROM segment: paddr=00020000h, vaddr=3c010000h, size=01140h (  4416) map
[esp32s3] [INF] IROM segment: paddr=00010000h, vaddr=42000000h, size=038AEh ( 14510) map`

Does anyone has a suggestion for this issue?
Thank you!

marekmatej, 2024-04-11:
> Hi Everyone, I also faced in to the same problem with ESP32-S3-DevKitC-1-N16R2. Just run the hello world sample but cannot flash to the board. The log is below: `SHA-256 comparison failed: [...]

Hi @bienxanh1901, is the `hello_world` sample not working at all? Or is your concerns about the messages? The messages are generated on the host whenever it sees the address that can be fetched from the elf. Can you please build the sample adding the shell subsystem and see if it works as expected? Use this build command: `west build -b esp32s3_devkitm samples/hello_world -p -DCONFIG_SHELL=y`

sylvioalves, 2024-04-11:
@bienxanh1901 Would you run again `west update` followed by `west blobs fetch hal_espressif`?