zephyrproject-rtos/zephyr #64954: DTLS server socket support multiple clients

pdgendt, 2023-11-08:
**Is your enhancement proposal related to a problem? Please describe.**
The current DTLS server socket implementation only supports a single client peer, because it has a single TLS context per socket file descriptor.
A side effect of this limitation is that, if the same host client initializes multiple handshakes on different ports, the server responds to all requests on the initial port.

**Describe the solution you'd like**
 An improvement would be to have a TLS context for each peer (`host:port` combination).

**Describe alternatives you've considered**
An idea I briefly looked into, is to split the `subsys/net/lib/sockets/sockets_tls.c` file into a TLS and a DTLS part, but nothing concrete.

**Additional context**
N/A

boaks, 2023-11-13:
> An improvement would be to have a TLS context for each peer (host:port combination).

Let me add, that using an implementation, which supports DTLS 1.2 Connection ID (as mbedTLS), it may be worth to support a TLS context based on the `CID` additional to `host:port`.

M1cha, 2025-10-08:
I think this should definitely be handled within the (D)TLS socket implementation because currently, you can enable DTLS on any UDP implemenation by changing the socket type and setting a couple socket-options.

DTLS client sockets actually have a similar issue: With raw UDP you just need a single socket which you can use to talk to multiple servers. With (D)TLS sockets that currently doesn't work, because there's just one tls-context internally.

As for using the host:port tuple vs connection IDs: The specification sounds like connection IDs are meant to update address:port information of the other side after it changed. While it may be possible to use it for multiplexing connections, it doesn't sound like is intended for that. I'd like to see feedback from security and/or DTLS experts regarding that.