zephyrproject-rtos/zephyr #64967: Bluetooth: Host: bt_att: ATT buffer re-use mechanism is controller dependent

andreeaDumitrache, 2023-11-08:
**Describe the bug**
The ATT buffer re-use mechanism relies on a controller sending number of completed packets after each TX to free the ATT TX metadata. Section 7.7.19 Number Of Completed Packets event of the Bluetooth Core specification (v5.4) states that "The rate with which this event is sent is manufacturer specific.". This means that a controller could choose to only send this event from time to time to ensure Host - Controller flow control, but not after each Tx. This would cause the Host to think that there is still an ongoing transaction on the ATT channel when it wants to send a new response, even if the previous response was already sent.

In bt_att_chan_create_pdu function:
```
if (re_use) {
        /* There can only ever be one transaction at a time on a
	        * bearer/channel. Use a dedicated channel meta-data to ensure
	        * we can always queue an (error) RSP for each REQ. The ATT
	        * module can then reschedule the RSP if it is not able to send
	        * it immediately.
	        */
        if (chan->rsp_meta.att_chan) {
	        /* Returning a NULL here will trigger an ATT timeout.
		        * This is better than an assert as an assert would
		        * allow a peer to DoS us.
		        */
	        LOG_ERR("already processing a REQ/RSP on chan %p", chan);
        
	        return NULL;
        }
		....
}
```

**To Reproduce**
I used Zephyr BLE on an internal platform, rebased on 3.5.0 tag. One board was flashed with the peripheral_ht sample and one with the central_ht sample. 2 calls in a row to bt_att_chan_create_pdu() without a HCI number of completed packets to free the ATT Tx metadata leads to erros. Service discovery is not finished properly and application functionality is affected.

**Related issues**
https://github.com/zephyrproject-rtos/zephyr/issues/64649
https://github.com/zephyrproject-rtos/zephyr/issues/62279

alwa-nordic, 2023-11-08:
Which controller are you using?

andreeaDumitrache, 2023-11-08:
> Which controller are you using?

I am using an internal NXP controller.

jori-nordic, 2023-11-16:
@andreeaDumitrache If the host can't (reliably) know what has and has not been sent on an ATT channel/bearer, how are we supposed to enforce the requirement that only one transaction shall be active at a time?

Does anyone have an idea on how to handle this rule?

alwa-nordic, 2023-11-16:
> @andreeaDumitrache If the host can't (reliably) know what has and has not been sent on an ATT channel/bearer, how are we supposed to enforce the requirement that only one transaction shall be active at a time?
> 
> Does anyone have an idea on how to handle this rule?

Then we can only enforce it from the point of view of the host. We must be able to accept a new request as soon as we have sent the response to the controller.

Elaborating on this: It's technically hard to define the point when sending of an HCI message is finished, so we should instead select an earlier point to complete the ATT transaction. E.g. When we start sending the HCI message. Or already when the response is placed on the TX-queue for the connection.

I don't think it's particularly important that we enforce correct behavior of the remote peer. There is wisdom in the [Robustness principle].

[Robustness principle]: https://en.wikipedia.org/wiki/Robustness_principle

jori-nordic, 2023-11-16:
@alwa-nordic so you're proposing that as soon as the whole ATT buffer has been clocked out through (one or multiple) `bt_send()`, the transaction is considered complete?
I think that would probably be a good enough solution, yeah. Let's see if there are other ideas.
Ideally, I would really like to remove `ATT_ENFORCE_FLOW`.

andreeaDumitrache, 2023-11-16:
> Then we can only enforce it from the point of view of the host. We must be able to accept a new request as soon as we have sent the response to the controller.
>
> Elaborating on this: It's technically [...]

I agree, I think enforcing flow control at ATT level is enough. For example, in chapter 3.3.3 Transaction of the 5.4 specification is stated that "On a server, a transaction shall start when a request is received by the server. A transaction shall complete when the response is sent by the server.". There is no mention of what is understood by "sent by the server", so it is not mandatory to have a strict confirmation from the Controller through a number of completed packets event that the response was sent to the peer. From the point of view of the Hos,t if it has sent the PDU to the Controller, its job is done. It is up to the Controller to take it from there and actually send it over the air.

PR #65272 (2023-11-17): Bluetooth: ATT: Revert re-use of RX buffer
The idea seemed good at the time :/

Reverting because:
- It doesn't work when `ATT_ENFORCE_FLOW=n`, and more controllers than I thought are actually using this. See #64967.
- It can result in a deadlock when not using `BT_RECV_BLOCKING` and the controller doesn't give events out of order. This is the case of the Softdevice controller
- It does create even more complexity, and I think we already have too much.

So instead, I will work on fixing the root cause, which is allowing the user to more finely control the allocation of buffers, in combination  [... cut]