zephyrproject-rtos/zephyr #64970: QEMU and twister support on windows not possible?

hf81-ble, 2023-11-08:
**Describe the bug**
I can build project with qemu (qemu_cortex_m3) 

1. Build Target
`west build -b qemu_cortex_m3 $ZEPHYR_BASE/samples/hello_world -p`

2. Run Target with QEMU
`west build -t run`

Please also mention any information which could help others to understand
the problem you're facing:
 - zepyhr 3.2.0

**To Reproduce**
Steps to reproduce the behavior:

If i now want to write a unit test with ztest and call twister to build and run the tests 

`python $ZEPHYR_BASE/scripts/twister -vvv -T MyTestSuit -p qemu_cortex_m3`

Build OK
RUN skipped

**Expected behavior**
Test should not only build, also run !

**Impact**
The tests are never running on windows hosts. Unit testing is essential for quality software development. ZTest is not usefull in that scenario. Tests should have an reason, why they are skipped to run. 

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Version 3.2.0

**Additional context**
Build and RUN from QEMU Targets seams not to be a problem. But running unit tests, which build seems also to work but not the RUN seams to me a broken feature which is not clearly commented neither wiki or in the docu itself: https://docs.zephyrproject.org/latest/develop/test/twister.html# If i read this, it seams promising to me, that unit tests also work in windows with qemu.

scripts/pylib/twister/twisterlib/testinstance line 192:

    # Global testsuite parameters
    def check_runnable(self, enable_slow=False, filter='buildable', fixtures=[], hardware_map=None):

        # running on simulators is currently not supported on Windows
        if os.name == 'nt' and self.platform.simulation != 'na':
            return False

Tests are prevented to run with twister to run windows! WHY not mention that serious impact!

If forcing over that, it seams to have an issue with named piped for qemu communication

´´´DEBUG   - adding xt-sim to default platforms
DEBUG   - Reading platform configuration files under C:\ncs\v2.3.0\zephyr\scripts\pylib\twister\boards...
DEBUG   - adding unit_testing to default platforms
DEBUG   - Reading platform configuration files under C:\ncs\v2.3.0\nrf\boards...
DEBUG   - platform filter: ['qemu_cortex_m3']
DEBUG   -   vendor filter: []
DEBUG   -     arch_filter: None
DEBUG   -      tag_filter: None
DEBUG   -     exclude_tag: None
INFO    - Building initial testsuite list...
INFO    - Writing JSON report C:\Projekte\CES\git\twister-out\testplan.json
INFO    - Adding tasks to the queue...
DEBUG   - adding qemu_cortex_m3\UnitTests\test.hf.ztest
INFO    - Added initial list of jobs to queue
DEBUG   - Launch process 0
DEBUG   - Launch process 1
DEBUG   - Launch process 2
DEBUG   - Launch process 3
DEBUG   - Launch process 4
DEBUG   - Launch process 5
DEBUG   - Launch process 6
DEBUG   - Launch process 7
Exception in thread qemu_cortex_m3\UnitTests\test.hf.ztest:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "C:/ncs/v2.3.0/zephyr\scripts/pylib/twister\twisterlib\handlers.py", line 836, in _thread
    out_fp, in_fp, log_out_fp = QEMUHandler._thread_open_files(
  File "C:/ncs/v2.3.0/zephyr\scripts/pylib/twister\twisterlib\handlers.py", line 784, in _thread_open_files
    os.mkfifo(fifo_in)
AttributeError: module 'os' has no attribute 'mkfifo'
´´´

nashif, 2023-11-28:
This is not supported and currently not planned, so it is not a bug.

closed 2023-11-28 by nashif

Dermiste, 2023-12-08:
Any idea if it will be supported in the near future? how much work we talking here?
I agree with OP that it should be mentioned in the doc that it won't run on windows although building & running with qemu is possible.

hf81-ble, 2023-12-08:
@Dermiste you can use the workaround (west build -t run) to build the test and run.....  What you dont have is the test case management to collect a test collection from different tests.... but in my case... we use this for libs or specific target and can manage them in one project. 

Problem is to export the test reports.... but for local tests it works. For CI pipelines we will setup an linux enviroment.

Dermiste, 2023-12-11:
Thanks that's what I figured too, we will also have a CI pipeline allowing a linux environment, in the meanwhile I'll just manage by just building and checking the output myself, I wish this would have just been mentioned in the doc.

digiexchris, 2024-01-22:
> This is not supported and currently not planned, so it is not a bug.

is there some specific incompatibility with Windows? considering qemu works with west, what's the blocker? I'm hoping to work around this but if there's code written to specifically skipping this platform I guess that won't work.