zephyrproject-rtos/zephyr #64972: shell.c does not provide mechanism to avoid data race

clamattia, 2023-11-08:
As per the [discussion](https://github.com/zephyrproject-rtos/zephyr/discussions/64632), `shell.c` does not currently provide a mechanisms to synchronize access to the prompt-string. Since the user is providing said string and the shell simply points to the users data directly, there is no way to avoid a data-race, when the `shell-thread` is reading and the user is writing to the prompt.

The problematic line: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/shell/shell.c#L1613

Suggestion would be to instead copy the prompt-string into a buffer private to the shell. For this operation the appropriate synchronization with the shell-thread can be done internally. This is also a more ituitiv api, since the prompt-buffer can live on the stack without invalidating any pointer in `shell.c`.

If instead the function is intended for use with string literals only, it would be desirable, that the documentation was clear on this and that for example a macro is provided, that asserts at compile-time, that only a string literal was used. 

For example code like this
```C
static char prompt[256];
strcpy(prompt, "Hello");
shell_prompt_change(sh, prompt);
```
is currently a data race, but seems perfectly fine.
It boils down to something like:
```C
char *a;
void shell_thread(void) {
    while(1) {
        // reads a
        if (a != NULL) {
            printf("%s\n", a);
        }
    }
}

void caller_thread(void) {
    static char prompt[256];
    strcpy(prompt, "Hello");
    a = prompt; // equivalent of shell_prompt_change(sh, prompt);
}
```

I wanted to share my idea on this but I am not 100% sure, so please correct me, if I am wrong.

jakub-uC, 2023-11-08:
Thank you for finding it. I will take care of this next week.

clamattia, 2023-11-27:
> Thank you for finding it. I will take care of this next week.

Has there been any news on this? Or was it postponed for priority reasons?

jakub-uC, 2023-11-29:
@clamattia  

A mutex is employed by the shell thread in the shell_process function, as indicated by k_mutex_lock(&sh->ctx->wr_mtx, K_FOREVER);. I can utilize this same mutex in the shell_prompt_change function. What do you think?

clamattia, 2023-11-29:
> A mutex is employed by the shell thread in the shell_process function, as indicated by sh->ctx->wr_mtx;. I can utilize this same mutex in the shell_prompt_change function. What do you think?

Whiles not knowing the details of how the `shell` is implemented, I do not think this is quite enough, since, the prompt is a pointer to user-provided memory. Which ever mutex you use, the user will not be able to lock it before reading/writing said memory and can therefore not guarantee, that there is no datarace. Or am I missing something?

IMO best would be to copy the prompt into internal memory on calls to `set_prompt` instead of pointing to user-provided memory and then additionally doing the measures necessary (mutexes etc...).

clamattia, 2023-11-29:
to illustrate locking like so is not enought IMO, because `caller_thread` can not lock before writing to `prompt`. Sure the transfer of the pointer `a` itself is ok and is done under the mutex. But writing to `prompt` itself can not.

```C
char *a;
struct mutex *b;
 
void shell_thread(void) {
    while(1) {
        // reads a
        lock(b);
        if (a != NULL) {
            printf("%s\n", a);
        }
        unlock(b);
    }
}

void caller_thread(void) {
    static char prompt[256];
    strcpy(prompt, "Hello");
    lock(b);
    a = prompt; // equivalent of shell_prompt_change(sh, prompt);
    unlock(b);
}
``

clamattia, 2023-11-29:
My proposal would be to instead:

```C
#define MAX_PROMT_SIZE xyz
char a[MAX_PROMPT_SIZE];
struct mutex *b;
 
void shell_thread(void) {
    while(1) {
        // reads a
        lock(b);
        printf("%s\n", a);
        unlock(b);
    }
}

void caller_thread(void) {
    static char prompt[256];
    strcpy(prompt, "Hello");
    lock(b);
    strncpy(a, promp, MAX_PROMPT_SIZE); // shell_prompt_change(sh, prompt); shell prompt should copy the prompt instead of pointing to it
    a[MAX_PROMPT_SIZE] = 0;
    unlock(b);
}
```

clamattia, 2023-11-29:
Let me know, if I was able to illustrate my thoughts.

jakub-uC, 2023-11-29:
Okay, now I understand.

Adding a private buffer is a solution, but it will again increase the shell memory usage, which people are already complaining about. I can add it under a configuration option, perhaps named 'safe prompt' or something similar.
What do you think?

jakub-uC, 2023-11-29:
On the other hand, maybe it doesn't make sense to do this as a configurable option. Let the length of the private buffer be configurable, but regardless, the RAM usage will be higher, yet this solution is much safer.

clamattia, 2023-11-29:
> Adding a private buffer is a solution, but it will again increase the shell memory usage, which people are already complaining about. I can add it under a configuration option, perhaps named 'safe prompt' or something similar. What do you think?

Understandable.
I would make the runtime-configureable, dynamic prompt always thread-safe but maybe make that itself configureable. Meaning, that the `shell_prompt_set`-function is not even available if one does not choose this feature.

Yes, I would not allow for compile-time configuration to choose safe vs non-safe code.

clamattia, 2023-11-29:
Probably most use cases are to set the prompt at compile time. So maybe it is enough to only make a safe `shell_prompt_set` available if the user explicitly wants a runtime configurable prompt and else force them to use  `CONFIG_..._PROMPT=` Kconfig option.

clamattia, 2024-01-29:
@jakub-uC My understanding is, that this will be fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/65943

clamattia, 2024-04-02:
> This issue has been marked as stale because it has been open (more than) 60 days with no activity. Remove the stale label or add a comment saying that you would like to have the label removed [...]

My understanding is, that this is being worked on still here: https://github.com/zephyrproject-rtos/zephyr/pull/65943

jakub-uC, 2024-04-03:
yes, there is one last CI test I need to fix.

PR #65943 (2024-04-04): shell: Add synchronization for prompt-string access in shell
Resolved a data race in shell.c by copying the user-provided prompt-string into a private buffer within the shell, ensuring proper synchronization with the shell-thread.

Fixes: #64972
commit f8263e82 (2024-04-04): shell: Add synchronization for prompt-string access in shell

Resolved a data race in shell.c by copying the user-provided
prompt-string into a private buffer within the shell, ensuring
proper synchronization with the shell-thread.

Fixes: #64972
closed 2024-04-04 by mmahadevan108