zephyrproject-rtos/zephyr #64985: nios2 msgdma smeaphore usage

teburd, 2023-11-08:
The nios2 msgdma DMA driver contributed by Intel uses a semaphore for synchronization around channel state transition and configuration functions. This is unlike most of the other drivers in the tree and the expectation is that all drivers in the tree at least adhere to a few simple rules around when dma calls are allowed and the what the behavior (in terms of state transitioning) is.

See https://github.com/zephyrproject-rtos/zephyr/pull/64982 to get a clearer picture of what the expectations of each DMA driver are

teburd, 2024-01-12:
I don't own nios2 and do not have time to resolve this issue for an architecture/platform I do not maintain, unassigned myself

nashif, 2024-01-29:
Planning to remove nios2 support after 3.6, so will not invest time on fixing a driver.

closed 2024-01-29 by nashif