zephyrproject-rtos/zephyr #65008: stm32: a single LPTIMER interrupt triggered in tickless mode after system is switched to SUSPEND_TO_IDLE state

r2r0, 2023-11-09:
**Describe the bug**
A nearly 1 second (about 940us) after system switches to SUSPEND_TO_IDLE state there is triggered a single LPTIMER interrupt.

 - MCU is STM32WB55
 - LPTIM is clocked from LSE (32768 Hz)
 - CONFIG_TICKLESS_KERNEL is enabled

**To Reproduce**
Steps to reproduce the behavior:
1. Build application with configuration as above and with some code (i.e. pin toggling) to detect LPTIMER ISR execution
2. Enter SUSPEND_TO_IDLE mode
3. See LPTIMER ISR execution trace

**Expected behavior**
In tickless kernel configuration and after system is suspended no LPTIMER interrupts should be generated.

**Impact**
Such behavior increases an average current consumption of the device 

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Version 3.5.0

**Additional context**

erwango, 2023-11-09:
@r2r0, could it be that LPTIM counter is not reset after the programmed IRQ ?
For instance:
T0: LPTIM counter rollover
T0 +1s: wakeup on k_sleep() expiry (on kernel programmed tick)
T0 +2s: LPTIM counter rollover.

Which can be avoided by reseting LPTIM counter on wakeup tick

r2r0, 2023-11-09:
@erwango For me it looks that above case starts when timer is no longer needed (condition  `ticks == K_TICKS_FOREVER` in the `sys_clock_set_timeout`). After that there is disabled control clock for LPTIMER but for some reason counter is still clocked (from LSE?) and eventually we get some kind of interrupt (probably autoreload match).

FRASTM, 2023-11-09:
> @erwango For me it looks that above case starts when timer is no longer needed (condition `ticks == K_TICKS_FOREVER` in the `sys_clock_set_timeout`). After that there is disabled control clock for [...]

Could this be completed by a call to `LL_LPTIM_Disable(LPTIM);`

FRASTM, 2023-12-13:
asking support to @andyross about that strange  `sys_clock_set_timeout(ticks, idle)` when passing _the special value K_TICKS_FOREVER_ When  CONFIG_TIMEOUT_64BIT,
The driver is comparing _ a  signed 32bit  ticks_ a   to K_TICKS_FOREVER to decide if it goes to sleep for ever in the line
` if (ticks == K_TICKS_FOREVER) ... sleep for ever` 
It comes from  k_msleep(K_TICKS_FOREVER) to the that sys_clock_set_timeout function and ticks is seen as 0x7FFFFFFF.
Because K_TICKS_FOREVER is seen as 0xFFFFFFFF , both values can never match !?

Does the sys_clock_set_timeout(int32_t ticks, bool idle)[](https://docs.zephyrproject.org/latest/kernel/services/timing/clocks.html#c.sys_clock_set_timeout)
cast the  K_TICKS_FOREVER on a signed 32bit  equal to 0x7FFFFFFF  (positive value)

FRASTM, 2023-12-13:
BTW why in [zephyr/include/zephyr/sys_clock.h ](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys_clock.h)

 * @brief Tick precision used in timeout APIs
 */
#ifdef CONFIG_TIMEOUT_64BIT
typedef **int**64_t k_ticks_t;
#else
typedef **uint**32_t k_ticks_t;
#endif

FRASTM, 2023-12-14:
> @FRASTM Do you have enabled `CONFIG_SYSTEM_SLOPPY_IDLE`? I am sorry, I forgot to add this to problem description. In case when `CONFIG_SYSTEM_SLOPPY_IDLE` is not set then system (`MAX_WAIT` in timeout.c) uses INT_MAX instead of -1 for "forever" value.

Weel in this case the driver should check

closed 2024-02-26 by aescolar