zephyrproject-rtos/zephyr #65018: microchip_mcp7940n_bbram_read ret value error

Yu3H0, 2023-11-09:
**Describe the bug**
when size = 0, the microchip_mcp7940n_bbram_read will return 0
```
static int microchip_mcp7940n_bbram_read(const struct device *dev, size_t offset, size_t size,
					 uint8_t *buffer)
{
	const struct microchip_mcp7940n_bbram_config *config = dev->config;
	struct microchip_mcp7940n_bbram_data *data = dev->data;
	size_t i = 0;
	int32_t rc = 0;

	if ((offset + size) > MICROCHIP_MCP7940N_SRAM_SIZE) {
		return -EINVAL;
	}

	k_mutex_lock(&data->lock, K_FOREVER);

	while (i < size) {
		LOG_DBG("Read from 0x%x", (MICROCHIP_MCP7940N_SRAM_OFFSET + offset + i));
		rc = i2c_reg_read_byte_dt(&config->i2c,
					  (MICROCHIP_MCP7940N_SRAM_OFFSET + offset + i),
					  &buffer[i]);

		if (rc != 0) {
			goto finish;
		}

		++i;
	}

finish:
	k_mutex_unlock(&data->lock);

	return rc;
}
```
but other handler like ```bbram_xec_read``` ```bbram_stm32_read``` and others will return -EINVAL;

**To Reproduce**
call the function microchip_mcp7940n_bbram_read

**Expected behavior**
when size is 0, the ret value should be -EINVAL;

Yu3H0, 2023-11-09:
so does microchip_mcp7940n_bbram_write

thedjnK, 2023-11-09:
I think you mean the bug is with the stm32 and xec drivers then, it is valid to read or write 0 bytes, it will do absolutely nothing but I don't see how that's an error in an input value. The function does exactly as described, you asked for 0 bytes, you got 0 bytes.

Yu3H0, 2023-11-09:
Hi, in function bbram_xec_read:
```
static int bbram_xec_read(const struct device *dev, size_t offset, size_t size,
			  uint8_t *data)
{
	const struct bbram_xec_config *dcfg = dev->config;

	if (size < 1 || offset + size > dcfg->size) { // check
		LOG_ERR("Invalid params");
		return -EFAULT;
	}

	bytecpy(data, dcfg->base + offset, size);
	return 0;
}
```
and function bbram_stm32_read
```
static int bbram_stm32_read(const struct device *dev, size_t offset, size_t size, uint8_t *data)
{
	const struct bbram_stm32_config *config = dev->config;
	uint32_t reg, begin, to_copy;

	if (size < 1 || offset + size > config->size) { //check
		return -EFAULT;
	}

	for (size_t read = 0; read < size; read += to_copy) {
		reg = STM32_BKP_REG(STM32_BKP_REG_INDEX(offset + read));
		begin = STM32_BKP_REG_BYTE_INDEX(offset + read);
		to_copy = MIN(STM32_BKP_REG_BYTES - begin, size - read);
		bytecpy(data + read, (uint8_t *)&reg + begin, to_copy);
	}

	return 0;
}
```
if the parameter size is zero, they return -EFAULT;
They all have the same upper level calls, so it gets weird, if the input parameters ```size``` are 0, some will return 0 and some will return -EFAULT.

thedjnK, 2023-11-09:
@jvasanth1 @erwango

erwango, 2023-11-10:
> I think you mean the bug is with the stm32 and xec drivers then, it is valid to read or write 0 bytes, it will do absolutely nothing but I don't see how that's an error in an input value. The function does exactly as described, you asked for 0 bytes, you got 0 bytes.

Mostly all drivers have the same behavior and return an error on 0.
Assigning to maintainer to check this.

commit 4efa11eb (2024-01-12): bbram: Add tests

Add emulation testing for it8xxx2, npcx, and mcp7940n. The test is made
to be generic and uses the backend API in order to verify both read and
write functionality. Additional tests are made for the API's limits when
reading/writing out of bounds.

Fixes #65018
closed 2024-01-12 by carlescufi