zephyrproject-rtos/zephyr #65023: RFC: DEFINE_DT_DEVICE_SUB for definition of sub-devices that do not have state or power management, as it is served by main device

de-nordic, 2023-11-09:
## Introduction

The goal of the proposal is to gather input and comments on addition of  `DEFINE_DT_DEVICE_SUB` themed macros that would allow to define sub-device, which is device defined within bounds of other device, and is a device that itself does not have power management nor state as all of that features are actually owned by the main device.
Sub-device here may be considered software construct that  is representing some functionality within a main device.
The proposal here links to PR that shows possible use and pinpoints current problems with implementations.

### Problem description

There are device that actually part of other device, for example soc-nv-flash compatible devices are defined, often, within flash-controller device and do not have own state, as their state depends on the state of controller.
Other examples of devices may be RTC devices that often combine RTS, battery backed SRAM and small amount of EEPROM, which all rely on the power state of the device but themselves have no such state.

Lack of support for such devices is confusing when attempting to using `DEVICE_DT_GET` or similar macros as you do not target the device you are trying to work with, but rather a parent, for example when accessing `flash0` labeled node that is `soc-nv-flash` compatible you have to do:
```
DEVICE_DT_GET(DT_PARENT(DT_NODELABLE(flash0)))
```
while when the same label is placed on SPI NOR, the invocation would look like:
```
DEVICE_DT_GET(DT_NODELABLE(flash0))
```

### Proposed change

Proposed change is to add subset of `DEFINE_DT_DEVICE` macros that would allow to define sub devices that defines no power management state and takes device state from parent/main device.

## Detailed RFC

Here I am providing source code https://github.com/zephyrproject-rtos/zephyr/pull/64089 that shows how such sub-device can be defined, with the soc-nv-flash compatible device being the target of the experiment.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

The change would require removing static from the definition of  `Z_DEVICE_STATE_DEFINE` to allow referencing it from the sub-device.

### Concerns and Unresolved Questions

The device does not have PM state and relies on the state of main device. This means that application may not be able to set power state of the sub device. The problem is that such device should not have own state as its state is the state of main device.
The problem with current implementation of PM is that to be able to pass PM calls from the sub-device to the main device, as should happen here, there is need for some significant `pm_device` type object to be attached to a device that does not support it at all, to just have a callback function that would be able to pass the PM actions to the main device.

There may be preliminary work required with moving the `action_cb` out of `struct pm_device` directly into `struct device`, and some rework of `pm_device_action_run` as the function does some basic checks on PM state object before calling the callback.

Note also that support for callback separate from the PM object may require a lot of changes to the code, as the PM callback would have to be provided via `DEVICE_DT_DEFINE` macros, which means that argument list for these macros needs to be expanded.

## Alternatives

Leaving things as they are. The problem with such approach is that often it is hard to add instances of devices, considered here sub-devices, as they are actually represented by the main device. The main device, as for example flash controller mentioned above, often exists as a single instance in DTS, so making it multiple instance device in driver would be rather bad solution.

tbursztyka, 2024-02-20:
This would require a thorough review of all possible use-cases to get so a relevant solution. It's an issue that has been here forever. It would be indeed nice to tackle this.
Anyway, you assume that it does not need PM, but I remember seeing devices that could have sub-devices with the ability to be put off/on.
Another issue that could perhaps exist is a driver for a device that can exist either as a standalone device or as a sub-device. Then we need a way (through DTS) to mitigate this.
There are other issues, but these are mostly due to the way things are done currently so not from your idea per-se.

de-nordic, 2024-02-22:
And the third thing I was trying to do but completely forgot about: I was trying to change soc-nv-flash into generic driver that would serve all soc-nv-flash devices by passing calls to parent. This way you could have multiple instance of the device, even with different controllers.