zephyrproject-rtos/zephyr #65025: [Backport v3.5-branch] Failed to backport #64873

zephyrbot, 2023-11-09:
This issue was created automatically because the backport of #64873 to `v3.5-branch` failed.

The backport to `v3.5-branch` failed:
```
Auto-merging drivers/flash/flash_mcux_flexspi_nor.c
CONFLICT (content): Merge conflict in drivers/flash/flash_mcux_flexspi_nor.c
error: could not apply 9dd8f94fd4... drivers: flash: mcux_flexspi_nor: Remove flash reads while programming
hint: After resolving the conflicts, mark them with
hint: "git add/rm <pathspec>", then run
hint: "git cherry-pick --continue".
hint: You can instead skip this commit with "git cherry-pick --skip".
hint: To abort and get back to the state before "git cherry-pick",
hint: run "git cherry-pick --abort".
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v3.5-branch v3.5-branch
# Navigate to the new working tree
cd .worktrees/backport-v3.5-branch
# Create a new branch
git switch --create backport-64873-to-v3.5-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x 9dd8f94fd47b8b399c230fac2c24e4949cc25b99
# Push it to GitHub
git push --set-upstream origin backport-64873-to-v3.5-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v3.5-branch
```
Then, create a pull request where the `base` branch is `v3.5-branch` and the `compare`/`head` branch is `backport-64873-to-v3.5-branch`.

dleach02, 2024-01-09:
@danieldegrasse, removing stale bot label

danieldegrasse, 2024-01-17:
Backport was done in https://github.com/zephyrproject-rtos/zephyr/pull/65028, this can be closed

closed 2024-01-17 by danieldegrasse