zephyrproject-rtos/zephyr #65036: Enable SoC flash controller on M4 of a STM32H7 SoC for STM32H747i-disco board

mkaricheri, 2023-11-09:
**Is your feature request related to a problem? Please describe.**

For a specific project, there should be flexibility to use either M4 or M7 for running Zephyr. For our project, we are using M4 for running zephyr. Currently SoC flash controller is not supported on M4. This request is to enable the support

**Describe the solution you'd like**

Basically SoC flash controller device should be able to allocate to M4. I have created a SoC flash test code which runs fine from M7. But when it is run on M4, the flash test program just hang. My changes are here https://github.com/zephyrproject-rtos/zephyr/compare/main...mkaricheri:zephyr:latest/stm32h747i-disco-soc-flash-m4 

Basically I would like to work on this and requiring some support from the community to understand what is the current limitation so that I can work to remove the same so that my sample works from M4 as well.

**Describe alternatives you've considered**

Unfortunately, there is no alternative since our project requires M4 to run Zephyr.

**Additional context**
Build it as 
west build -p always -b stm32h747i_disco_m4 samples/drivers/soc_flash_stm32h7 -- -DOVERLAY_CONFIG=boards/stm32h747i_disco_m4.conf -DDTC_OVERLAY_FILE=boards/stm32h747i_disco_m4.overlay

I also build a hello world application and run it on M7 as

west build -p always -b stm32h747i_disco_m7  samples/hello_world

Here is the log I get when run the soc flash test application

*** Booting Zephyr OS build zephyr-v3.5.0-1189-g8e42f489f052 ***

   STM32H7 Flash Testing
   =====================
   Available partitions on SoC Flash
   image1_partition        @0, size 786432
   scratch_partition       @c0000, size 262144

Test 1: Flash erase page at 0xc0000, size 262144

mkaricheri, 2023-11-09:
@erwango could you tell me what is the current limitation of running the SoC flash controller driver on M4. I have written a SoC Flash test application similar to soc_flash_nrf sample, and it runs happily on M7. I need to get this working on M4 and would need your help to understand the above limitation so that I can work to resolve the same. 

Thanks in advance for your time on this request.

Murali

mkaricheri, 2023-11-09:
On M7 my test application runs fine and gives the following log
*** Booting Zephyr OS build zephyr-v3.5.0-1188-g86c1c0860d5a ***

   STM32H7 Flash Testing
   =====================
   Available partitions on SoC Flash
   image1_partition        @0, size 1048576
   image2_partition        @100000, size 786432
   scratch_partition       @1c0000, size 262144

Test 1: Flash erase page at 0x1c0000, size 262144
   Flash erase succeeded!

Test 2: Flash write
   Wrote 1024 blocks of 32 bytes
   Wrote 2048 blocks of 32 bytes
   Wrote 3072 blocks of 32 bytes
   Wrote 4096 blocks of 32 bytes
   Wrote 5120 blocks of 32 bytes
   Wrote 6144 blocks of 32 bytes
   Wrote 7168 blocks of 32 bytes
   Wrote 8192 blocks of 32 bytes
   read and verified upto 0x1c8000
   read and verified upto 0x1d0000
   read and verified upto 0x1d8000
   read and verified upto 0x1e0000
   read and verified upto 0x1e8000
   read and verified upto 0x1f0000
   read and verified upto 0x1f8000
   8192 blocks read and verified correctly

Test 3: Page layout API
   Offset  0x001e0004:
     belongs to the page 15 of start offset 0x001e0000
     and the size of 0x00020000 B.
   Page of number 15 has start offset 0x001e0000
     and size of 0x00020000 B.
     Page index resolved properly
   SoC flash consists of 16 pages.

Test 4: Write block size API
   write-block-size = 32

FRASTM, 2023-11-10:
earlier :   https://github.com/zephyrproject-rtos/zephyr/issues/44783

mkaricheri, 2023-11-10:
@FRASTM thanks for the history, My debug also pointed to exception in LL_GetFlashSize() as mentioned in the above thread. Since this is a M7 only register, one option would be to define a KConfig option to set it. If not, driver go and check the value from the hardware. For M4 we could define the KConfig option value.

FRASTM, 2023-11-13:
Did you try to bypass the ART Flash cache accelerator, just for testing ?

mkaricheri, 2023-11-13:
> Did you try to bypass the ART Flash cache accelerator, just for testing ?

> ART Flash cache

How do we disable it? I will check if there is a CONFIG option to disable

FRASTM, 2023-11-14:
> > Did you try to bypass the ART Flash cache accelerator, just for testing ?
> 
> > ART Flash cache
> 
> How do we disable it? I will check if there is a CONFIG option to disable

It is directly configureed in the stm32h7_m4_init() of https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/st_stm32/stm32h7/soc_m4.c

mkaricheri, 2023-11-14:
@FRASTM Thanks for the information. I will try this. Do you know the current design details of this driver?
I would like to see the following supported.
1. Either Full or Half Bank shall be allocated to Each core. i.e M4 or M7 can get 2MiB of flash. In the half case, M7 will control Bank 0 and M4 Bank 1. I see that the number of sectors (16) is even though M7 DT node defines only half of the Flash (1MiB) allocated which seems wrong for me. It should be 8 since sector size is 128KiB.
2. Driver on M7 co-exists with driver on M4 when each is assigned half of the flash device. Looks like design takes care of this as it uses a hardware semaphore to provide locking.

So the first question I have is if M7 support assumes Half of the flash assigned to it or full. I had assigned full to M7 and it seems to work for me. So the half case needs to be supported and the number of sectors to be adjusted to 8 in this case. What do you think?

mkaricheri, 2023-11-14:
> > > Did you try to bypass the ART Flash cache accelerator, just for testing ?
> > 
> > 
> > > ART Flash cache
> > 
> > 
> > How do we disable it? I will check if there is a CONFIG option to disable
> 
> It is directly configured in the stm32h7_m4_init() function of https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/arm/st_stm32/stm32h7/soc_m4.c

It didn't make any difference. This is what I have tried
```
diff --git a/soc/arm/st_stm32/stm32h7/soc_m4.c b/soc/arm/st_stm32/stm32h7/soc_m4.c
index e2fe895511..4a22804ce6 100644
--- a/soc/arm/st_stm32/stm32h7/soc_m4.c
+++ b/soc/arm/st_stm32/stm32h7/soc_m4.c
@@ -33,9 +33,9 @@
 static int stm32h7_m4_init(void)
 {
        /* Enable ART Flash cache accelerator */
-       LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_ART);
+       /* LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_ART); */
        LL_ART_SetBaseAddress(DT_REG_ADDR(DT_CHOSEN(zephyr_flash)));
-       LL_ART_Enable();
+       /* LL_ART_Enable(); */

        /* In case CM4 has not been forced boot by CM7,
         * CM4 needs to wait until CM7 has setup clock configuration
```

mkaricheri, 2023-11-14:
> > @FRASTM branch is now working. I found an offset issue in get_sector() function when the flash device is bank 2 which is the case with M4 owning bank 2.
> 
> Here is the log from my flash driver example
> 
> ```
> *** Booting Zephyr OS build zephyr-v3.5.0-1189-g4efb3510572d ***
> 
>    STM32H7 Flash Testing
>    =====================
>    Available partitions on SoC Flash
>    image1_partition        @0, size 786432
>    scratch_partition       @c0000, size 262144
> 
> Test 1: Flash erase page at 0xc0000, size 262144
>    Flash erase succeeded!
> 
> Test 2: Flash write
>    Wrote 1024 blocks of 32 bytes
>    Wrote 2048 blocks of 32 bytes
>    Wrote 3072 blocks of 32 bytes
>    Wrote 4096 blocks of 32 bytes
>    Wrote 5120 blocks of 32 bytes
>    Wrote 6144 blocks of 32 bytes
>    Wrote 7168 blocks of 32 bytes
>    Wrote 8192 blocks of 32 bytes
>    read and verified upto 0xc8000
>    read and verified upto 0xd0000
>    read and verified upto 0xd8000
>    read and verified upto 0xe0000
>    read and verified upto 0xe8000
>    read and verified upto 0xf0000
>    read and verified upto 0xf8000
>    8192 blocks read and verified correctly
> 
> Test 3: Page layout API
>    Offset  0x000e0004:
>      belongs to the page 7 of start offset 0x000e0000
>      and the size of 0x00020000 B.
>    Page of number 6 has start offset 0x000c0000
>      and size of 0x00020000 B.
>      Page index resolved properly
>    SoC flash consists of 16 pages.
> 
> Test 4: Write block size API
>    write-block-size = 32
> ```

One issue is the flash write is very slow from M4.

mkaricheri, 2023-11-14:
@FRASTM branch is now working. I found an offset issue in get_sector() function when the flash device is bank 2 which is the case with M4 owning bank 2.

mkaricheri, 2023-11-14:
> @FRASTM branch is now working. I found an offset issue in get_sector() function when the flash device is bank 2 which is the case with M4 owning bank 2.

Here is the log from my flash driver example
```
*** Booting Zephyr OS build zephyr-v3.5.0-1189-g4efb3510572d ***

   STM32H7 Flash Testing
   =====================
   Available partitions on SoC Flash
   image1_partition        @0, size 786432
   scratch_partition       @c0000, size 262144

Test 1: Flash erase page at 0xc0000, size 262144
   Flash erase succeeded!

Test 2: Flash write
   Wrote 1024 blocks of 32 bytes
   Wrote 2048 blocks of 32 bytes
   Wrote 3072 blocks of 32 bytes
   Wrote 4096 blocks of 32 bytes
   Wrote 5120 blocks of 32 bytes
   Wrote 6144 blocks of 32 bytes
   Wrote 7168 blocks of 32 bytes
   Wrote 8192 blocks of 32 bytes
   read and verified upto 0xc8000
   read and verified upto 0xd0000
   read and verified upto 0xd8000
   read and verified upto 0xe0000
   read and verified upto 0xe8000
   read and verified upto 0xf0000
   read and verified upto 0xf8000
   8192 blocks read and verified correctly

Test 3: Page layout API
   Offset  0x000e0004:
     belongs to the page 7 of start offset 0x000e0000
     and the size of 0x00020000 B.
   Page of number 6 has start offset 0x000c0000
     and size of 0x00020000 B.
     Page index resolved properly
   SoC flash consists of 16 pages.

Test 4: Write block size API
   write-block-size = 32

```

mkaricheri, 2023-11-14:
@erwango @FRASTM Do you think the commits on this branch is ready for review? If so, I would like to send a PR for the same. Otherwise, please give a feedback so that I can incorporate them before sending the PR. Thanks.

mkaricheri, 2023-11-14:
This is the branch I have rebased to latest on main https://github.com/zephyrproject-rtos/zephyr/compare/main...mkaricheri:zephyr:latest/stm32h747i-disco-soc-flash-m4

erwango, 2023-11-15:
> @erwango @FRASTM Do you think the commits on this branch is ready for review? If so, I would like to send a PR for the same. Otherwise, please give a feedback so that I can incorporate them before sending the PR. Thanks.

You can submit your PR in Draft mode and we'll discuss from there.

FRASTM, 2023-11-15:
> @FRASTM branch is now working. I found an offset issue in get_sector() function when the flash device is bank 2 which is the case with M4 owning bank 2.

good catch

mkaricheri, 2023-11-15:
> > @erwango @FRASTM Do you think the commits on this branch is ready for review? If so, I would like to send a PR for the same. Otherwise, please give a feedback so that I can incorporate them before sending the PR. Thanks.
> 
> You can submit your PR in Draft mode and we'll discuss from there.

I have created a draft PR for this @erwango  cc: @FRASTM

erwango, 2023-11-16:
@mkaricheri can you provide the reference here ? (I can find it, but could be useful for others and it's good practice in general)

FRASTM, 2023-11-16:
https://github.com/zephyrproject-rtos/zephyr/pull/65250

mkaricheri, 2023-11-17:
@erwango https://github.com/zephyrproject-rtos/zephyr/pull/65250

FRASTM, 2024-07-17:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/74121

closed 2024-09-18 by erwango