zephyrproject-rtos/zephyr #65062: Cannot set Media Player Name and Track Title

piotrnarajowski, 2023-11-10:
**Describe the bug**
Two GMCS Server role test cases require updating values of Media Player Name characteristic or Track Title characteristic. Currently Media Player is initialized with hardcoded Media Player Name and various groups of tracks. There is no API that allows changing values of these characteristics. 

**To Reproduce**
Can be reproduced with AutoPTS.

**Expected behavior**
API should allow to change values of Media Player Name and Track Title characteristic.

**Impact**
Test case verdict is inconclusive.

**Logs and console output**
[logs.zip](https://github.com/zephyrproject-rtos/zephyr/files/13319695/Zephyr.errata.logs.zip)

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain: Zephyr SDK 0.16.3
 - #64377

Thalley, 2023-11-13:
Modified this to be an enhancement as it is not a bug, simply a missing feature.

asbjornsabo, 2023-11-28:
I do not think I agree to this request.
A media player will have a given name (e.g. Spotify).  I can not see that it makes sense to have an API to change that name.  Ditto for the Icon Object ID and the Icon URL, which also are given for a given player.
There are test commands to trigger callbacks for notifying that Player Name, Icon Object ID and Icon URL have been changed, though - see mpl.c in the shell commands.  That used to be sufficient for testing.

What could make sense is to extend the current GMCS implementation to support multiple players.  In that case, there is a need to register the name of the new player.  (But that would also imply having a complete set of all the other characteristics, or to switch all the characteristics over to new values.)  This goes along with implementing proper MCS support - there may be multiple players with separate instances, GMCS will point to one of them (the "current" one).

The player name is only "soft" hardocded - it is a Kconfig option.
The groups and tracks are hard coded, as the media player implementation is so far only a simple mock-up.

asbjornsabo, 2023-11-28:
As for Track Title: Is the request to change the title of the track (that is not specified)?  Or is it to change to another track with another title?
Track Title, as Player Name, are Read only.  The server can not expect to change them.

piotrnarajowski, 2023-11-28:
@asbjornsabo the instruction in MMI is: "Please update Media Player Name/Track Title characteristic with value whose length is greater than the (ATT_MTU-3). Click OK when it is ready.
Test specification: "The Upper Tester commands the IUT to update the characteristic specified by the Characteristic UUID referenced in Table 4.19 (media player name, track title) with a different value whose length is greater than the (ATT_MTU-3)."

asbjornsabo, 2023-11-28:
OK - now I see.
These tests are for setting "oversize" values and changing the name while in a long read.
I guess that must be added, but it should be with the other "test" functions - this is not something that should be in the public API.

Thalley, 2023-11-28:
> I can not see that it makes sense to have an API to change that name. 

We may not see an immediate usecase for it, but that doesn't mean it isn't there. In any case, the characteristic is (optionally) notifiable, which indicates that it can be notified, and thus changed. 

One example of where this may happen, is something I see in my Linux installation. When I have some media playing from my Firefox browser, it will have the name similar to "Firefox - Youtube" or "Firefox - Spotify" - So even though Firefox is _technically_ the media service, the underlying service may change, and thus the service name changes.

Thalley, 2026-08-15:
We probably should wait with this until https://github.com/zephyrproject-rtos/zephyr/issues/57441 and https://github.com/zephyrproject-rtos/zephyr/issues/103830 have been implemented and merged

Thalley, 2026-09-20:
Closing as fixed by https://github.com/zephyrproject-rtos/zephyr/pull/118253
There shouldn't be an API to change track titles

closed 2026-09-20 by Thalley