zephyrproject-rtos/zephyr #65069: Add flag to distinguish short / long button press for longpress device

deveritec-rosc, 2023-11-10:
Today I started using the `zephyr,input-longpress` as introduced in #55787 .
While using, I found it cumbersome to have to define
- the input code
- the short code
- the long code

While this makes sense in a way that input-longpress enhances the input subsystem, assigning different keys to "short" and "long" just to be able to distinguish feels cumbersome.

Proposal:
- reserve the 4 MSB of `struct input_event::code` for additional information
- add defines which hold the information (e.g. "short" or "long") to `input-event-codes.h`
- optional: add macro which puts arbitrary code in the upper 4 bits

Alternative:
- expanding `struct input_event`:
needs more memory, and `code` has enough space to store additional information
- auto-apply information: would break existing implementations

fabiobaltieri, 2023-11-10:
Hi @deveritec-rosc, the subsystem is heavily inspired by the Linux one where those codes are pretty much opaque, and I think that's hot it should stay. This allows the application developer to use the codes with whatever logic they want and ensure that the subystem does not get in the way, I'd suggest you to use that instead and define your own convention as you like and use it on your app. From what I understand you may want to define your application specific `DT_...` macro to set whatever bits you want to whatever meaning you want and then use it on your application code, and that's perfectly fine, but at subsystem level a code is a code it it should not be messed up with.

deveritec-rosc, 2023-11-10:
Okay, also fine with me :)

closed 2023-11-10 by deveritec-rosc