zephyrproject-rtos/zephyr #65099: I3C: reporting earlier than expected EoD or CCC with different sizes

XenuIsWatching, 2023-11-12:
# Description
With I3C, when a controller does a read whether, through a private transfer or a CCC, the target is responsible for issuing the End of Data through the T bit, but the way the API is currently implemented, there is no way to communicate this back up. For example, if a controller tries to read 100 Bytes (where the controller is to do an abort after 100 (or 101 bytes)) from a Target and the Target gives the EoD at the 42nd Byte, then there is no way to currently know where the EoD occurred and how far it can be read up to in the provided buffers.

This can also present a problem when trying to perform a CCC MXDS which the format used is defined based on the number of bytes returned by the Target. There needs to be a way for the length to be communicated back up.

# Proposed Solution
One Solution would be for the driver to overwrite the data length of the packet passed to it for the application then to read and know if the data ended earlier than expected, but this just ""feels"" kinda dirty. As the application would always need to store another copy of the data length it just first passed into the API for it to compare if it was modified after it returns.

dcpleung, 2023-11-13:
I think adding another field to `i3c_msg` would probably be the best, as it avoids breaking current users.

dcpleung, 2023-11-13:
@XenuIsWatching I am thinking adding a `num_xfer` in `i3c_msg`. The MCUX driver is easy to set it. However, the Cadence driver is a bit complicated. Could you make a patch on the Cadence driver so I can combine them into one PR? Or, if you wish, you can do all the needed changes.

XenuIsWatching, 2023-11-13:
> @XenuIsWatching I am thinking adding a `num_xfer` in `i3c_msg`. The MCUX driver is easy to set it. However, the Cadence driver is a bit complicated. Could you make a patch on the Cadence driver so I can combine them into one PR? Or, if you wish, you can do all the needed changes.

I made a PR with it all https://github.com/zephyrproject-rtos/zephyr/pull/65140

closed 2023-12-04 by henrikbrixandersen