zephyrproject-rtos/zephyr #65109: RFC: API change: drivers: can: remove CAN_FILTER_FDF flag

henrikbrixandersen, 2023-11-13:
## Introduction
The `CAN_FILTER_FDF` was added in #52483 on the premise that the NXP S32 CANEXCEL driver (added in #52450) needed to know whether a CAN filter should match a classic CAN or a CAN FD frame; it could not match both.

This was opposite of all other in-tree supported CAN controllers, which are unable to filter on classic CAN/CAN FD in hardware. In order to maintain a uniform behaviour across the in-tree CAN FD capable CAN controller drivers, support for this flag was added to these by amending their hardware filtering with a software-based filter, which would discard frames not matching the presence/non-presence of the `CAN_FILTER_FDF` flag.

### Problem description
Unfortunately, this added software-based filtering (or rather, the mix of hardware- and software-based filtering) leads to issues like the one reported in #64554. 

Investigation of how this flag was used in the NXP S32 CANEXCEL IP core revealed that the premise presented in #52483 was false. The NXP S32 CANEXCEL IP core has the ability to filter on classic/FD vs. CAN XL frames, but not between classic CAN and CAN FD frames. The NXP HAL API `Canexcel_Ip_SetRxIndividualMask ()` leads to this false conclusion (see https://github.com/zephyrproject-rtos/hal_nxp/blob/78d1912dbc2b5f6e114d7fd19a31053716a3b01d/s32/drivers/s32ze/Can_CANEXCEL/include/CanEXCEL_Ip.h#L93C48-L93C48), while in fact the implementation only distinguishes between CAN XL frames/non-XL frames (see https://github.com/zephyrproject-rtos/hal_nxp/blob/78d1912dbc2b5f6e114d7fd19a31053716a3b01d/s32/drivers/s32ze/Can_CANEXCEL/src/CanEXCEL_Ip.c#L297). This is in line with the reference manual for the CANEXCEL IP core.

### Proposed change
The `CAN_FILTER_FDF` flag should never have been introduced. It leads to overhead, false promises of the API, and issues like those reported in #64554. I propose this flag is removed again as no known CAN controllers provide hardware support for filtering between classic CAN and CAN FD frames.

## Detailed RFC
Please see #65108

### Proposed change (Detailed)
Please see #65108

### Dependencies
The detailed RFC currently includes the proposed changes from #65058 in order to completely remove the mix of hardware- and software-based filtering from the Bosch M_CAN driver backend.

### Concerns and Unresolved Questions
None.

## Alternatives
None.

henrikbrixandersen, 2023-11-13:
@manuargue, @congnguyenhuu, @dleach02, @str4t0m, @martinjaeger, @alexanderwachter: FYI

martinjaeger, 2023-11-13:
@henrikbrixandersen Your explanations make sense and I agree to remove the software filtering based on FD flag.

The FDF bit in the CAN frame is not part of the arbitration field, but belongs to the control field. So it does not make much sense to filter based on the FDF flag either.

henrikbrixandersen, 2023-11-14:
@manuargue, @congnguyenhuu: Any opinions on this?

manuargue, 2023-11-15:
> @manuargue, @congnguyenhuu: Any opinions on this?

@henrikbrixandersen based on my general knowledge of CAN and the evidence presented here, it seems to me your proposal is the right move. I have no objections.

congnguyenhuu, 2023-11-15:
> @manuargue, @congnguyenhuu: Any opinions on this?
Hello,
You are right. NXP S32 CANEXCEL not ability to set filter on classic/FD frames by `Canexcel_Ip_SetRxIndividualMask()`. It is only mask to filter msg ID.
But We need to configure type frame for each Rx msg to receive corresponding frames classic/FD frames with [Canexcel_Ip_ConfigRx()](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/s32/drivers/s32ze/Can_CANEXCEL/src/CanEXCEL_Ip.c#L311C24-L311C44)
Need to configure `CTRL.RXCTRL`, configure allows only receive classic/FD frames.
For FD frame:
[CANEXCEL.EXL_MSGD[instance]->MSGDSC[descNo].CTRL.RXCTRL |= CANXL_MSG_DESCRIPTORS_RXCTRL_RXFDFRM_MASK;](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/s32/drivers/s32ze/Can_CANEXCEL/src/CanEXCEL_Ip.c#L359)

=>So NXP S32 CANEXCEL still need the `FDF` flag for `Canexcel_Ip_ConfigRx`

henrikbrixandersen, 2023-11-15:
> Hello, You are right. NXP S32 CANEXCEL not ability to set filter on classic/FD frames by `Canexcel_Ip_SetRxIndividualMask()`.
> 
> But the current shim driver for CANEXCEL uses Receive Message Descriptor mode for receiving (beside that hardware also support RxFIFO mode). In that mode, there is Message Descriptor Control which configures the Receive Message Descriptor. The bit `RXFDFRM` in `CTRL.RXCTRL` register indicates whether the message descriptor is configured to receive classic or FD type frame.
> 
>     * 0 for classic frame
> 
>     * 1 for FD frame
> 
> 
> It is used to filter type frame beside filter by message ID mask. =>So NXP S32 CANEXCEL still need the `FDF` flag for to know the type frame before sending/receiving. The type frame is configured by [Canexcel_Ip_ConfigRx()](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/s32/drivers/s32ze/Can_CANEXCEL/src/CanEXCEL_Ip.c#L311C24-L311C44)

Okay. Thank you for the analysis. The driver will need to be changed to RX FIFO mode, then, in order to get rid of this flag?

congnguyenhuu, 2023-11-15:
> Okay. Thank you for the analysis. The driver will need to be changed to RX FIFO mode, then, in order to get rid of this flag?

I need more time to analyze this mode. I'm not even sure whether it fits the expectation of CAN API Zephyr.
Your proposal will break current support CAN FD for CANEXCEL shim driver.

congnguyenhuu, 2023-11-20:
> I need more time to analyze this mode. I'm not even sure whether it fits the expectation of CAN API Zephyr. Your proposal will break current support CAN FD for CANEXCEL shim driver.

I'm quite busy right now so I can't come to a conclusion right away. I will back to this early next month.

congnguyenhuu, 2023-12-10:
> > > Okay. Thank you for the analysis. The driver will need to be changed to RX FIFO mode, then, in order to get rid of this flag?

Hello @henrikbrixandersen, after investigating I see that using RxFIFO for NXP S32 CANEXCEL, it doesn't require this flag and also fits the expectation of CAN APIs.
So I will add RxFIFO support for NXP S32 CANEXCEL.
Could you please open PR for apply your proposal ? Thanks.

henrikbrixandersen, 2023-12-10:
> Hello @henrikbrixandersen, after investigating I see that using RxFIFO for NXP S32 CANEXCEL, it doesn't require this flag and also fits the expectation of CAN APIs.
> So I will add RxFIFO support for NXP S32 CANEXCEL.
> Could you please open PR for apply your proposal ? Thanks.

Great! You should be able to apply your CANEXCEL changes on top of #65108. We can pick and squash the relevant commits afterwards.

henrikbrixandersen, 2024-01-02:
@congnguyenhuu Any news regarding this?

henrikbrixandersen, 2024-01-02:
Related RFC for removing run-time RTR filtering suffering from similar issues: https://github.com/zephyrproject-rtos/zephyr/issues/67128

congnguyenhuu, 2024-01-03:
> @congnguyenhuu Any news regarding this?

Perhaps this weekend, I will open PR for this.

congnguyenhuu, 2024-01-03:
> @congnguyenhuu Any news regarding this?

Hello, Perhaps this weekend, I will open PR for this.

congnguyenhuu, 2024-01-06:
Hello @henrikbrixandersen, I opened this draft PR https://github.com/zephyrproject-rtos/zephyr/pull/67264.
Could you please help me review it.
Thanks

closed 2024-01-19 by henrikbrixandersen