zephyrproject-rtos/zephyr #65118: Non-XIP (ARM) images lock up when common malloc area Kconfig is set to -1

nordicjm, 2023-11-13:
**Describe the bug**
Testing MCUboot's RAM LOAD mode seems to fail when trying to boot an image whereby the common malloc size is -1 (this means to use the rest of available RAM for heap). Upon setting this Kconfig to 0 or another value (e.g. 4096) then the image loads and functions fine.

**Expected behavior**
Image to boot and work

**Impact**
Showstopper

**Logs and console output**
There is no output at all, not even the boot banner outputs, trying to debug it (before knowing it was the malloc Kconfig) did not yield anything useful, the kernel seemed to start then get to a point whereby the stack trace made no sense and was corrupt.

**Environment (please complete the following information):**
 - Commit SHA or Version used: 3003cc1938f53074ad717af79d1ff1e7ff15b514

keith-packard, 2023-11-13:
Can you provide a .config for the failing build? I suspect it's an issue with the value of the symbols used to compute the amount of ram remaining after startup. The code uses `CONFIG_SRAM_BASE_ADDRESS`, `CONFIG_SRAM_SIZE`, and `&_end`.

nordicjm, 2023-11-13:
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/13335136/config.txt)
Attached
[map.txt](https://github.com/zephyrproject-rtos/zephyr/files/13335137/map.txt)
Also attached map file

keith-packard, 2023-11-13:
Hrm. Maybe just not use -1 mode by default without XIP for now? I don't know how to reproduce the problem.

nordicjm, 2023-11-13:
When https://github.com/zephyrproject-rtos/zephyr/pull/65131 is merged, I can provide instructions for generating a build for a board that will demonstrate the issue

Laczen, 2023-11-13:
@nordicjm, have you set the `CONFIG_BOOTLOADER_SRAM_SIZE` to a correct value ? According to the 3.5 migration guide bootliaders that use a part of ram should set this to a correct value.

nordicjm, 2023-11-14:
> @nordicjm, have you set the `CONFIG_BOOTLOADER_SRAM_SIZE` to a correct value ? According to the 3.5 migration guide bootliaders that use a part of ram should set this to a correct value.

Here's the thread whereby the person that originally added it was asked about it and did not respond, I had forgotten it's seemingly only used by 1 board: https://github.com/zephyrproject-rtos/zephyr/pull/60371

Therefore I will post a PR to just remove it entirely, the same can be done using DTS, the single user of this is not responsive and it's for a board with no public documentation at all

closed 2024-06-16 by github-actions[bot]