zephyrproject-rtos/zephyr #65121: RFC: Remove CONFIG_BOOTLOADER_SRAM_SIZE

nordicjm, 2023-11-13:
## Introduction

This is an RFC to discuss potentially removing of the Kconfig BOOTLOADER_SRAM_SIZE symbol, why you might ask, read on to find out why:

### Problem description

This symbol sounds like it's useful for non-XIP applications, so you can have your application set up then set this symbol to the amount of RAM that your bootloader uses and away you go, thinking it would offset the start of your image by the value given. This however, is not the case, this is the Cortex-M code for this Kconfig:
```
#if defined(CONFIG_XIP)  
#if defined(CONFIG_IS_BOOTLOADER)
#define RAM_SIZE (CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR (CONFIG_SRAM_BASE_ADDRESS + \
        (CONFIG_SRAM_SIZE * 1K - RAM_SIZE))
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif
#else
#define RAM_SIZE (CONFIG_SRAM_SIZE * 1K - CONFIG_BOOTLOADER_SRAM_SIZE * 1K)
#define RAM_ADDR CONFIG_SRAM_BASE_ADDRESS
#endif
```

So from looking at it, I don't really see what the point of this symbol is, it adjusts the total RAM size of your application, but does nothing to the base SRAM address? If you assume that the bootloader uses 64KB of RAM (at the start address of RAM, as we do other partitions e.g. flash) then want to load your application after, you can't use this Kconfig, you have to play with the Kconfigs for the RAM base address and total RAM size.

### Proposed change

Remove this Kconfig. Alternatively, change it so it applies to the start of RAM and affects both the address and size, instead of just the end size address.

### Concerns and Unresolved Questions

Unknown is anyone using this Kconfig, whom that is I do not know. @GAnthony since you seem to use this symbol on cc3220sf, do you have any input on this? And why it adjusts the end address of RAM to the application, not the start address and total size?

Laczen, 2023-11-13:
@nordicjm, at first I was thinking that this was wrong, but I think it is correct. It supposes the bootloader runs in xip, and assigns the a region at the highest part to the bootloader.

nordicjm, 2023-11-13:
> @nordicjm, at first I was thinking that the code was wrong, but I think it is correct. It supposes the bootloader runs in xip, and assigns a ram region at the highest part to the bootloader.

Seems like moving to partitioning the RAM in dts would be a solution

Laczen, 2023-11-13:
Why. there is no problem with the existing solution. The application is always setup to the start of the ram, only a modification is applied to the bootloader. It would even be possible to reclaim the bootloader ram if there is no application written to the bootloader (ram) part.

nordicjm, 2023-11-13:
We use DTS for configuration, not Kconfig.

PR #65156 (2024-01-08): kconfig: Deprecate BOOTLOADER_SRAM_SIZE
    This Kconfig is seemingly unused with exception to a single board,
    whose maintainers have been unresponsive for a long time. It is
    not useful to any other board and can be set in dts, therefore
    deprecate it and schedule removal after the Zephyr 3.7 release

Fixes #65121
closed 2024-01-08 by carlescufi