zephyrproject-rtos/zephyr #65123: subsys: llext: llext.simple.arm fails

katgiadla, 2023-11-13:
**Describe the bug**
The test `llext.simple.arm` fails.

Observed for:
- `nrf52dk_nrf52832`
- `nrf52840dk_nrf52840`
- `nrf5340dk_nrf5340_cpuapp`
- `nrf5340dk_nrf5340_cpuapp_ns`
- `nrf5340dk_nrf5340_cpunet`
- `nrf9160dk_nrf9160`
- `nrf9160dk_nrf9160_ns`

**To Reproduce**
Steps to reproduce the behavior:

1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/llext -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
For  `nrf5340dk_nrf5340_cpuapp_ns` and `nrf9160dk_nrf9160_ns` (the logs shows in the loop until timeout):
```
ype 3 sym_bind 0 sym_ndx 5[0m
[00:00:00.945,129] [0m<inf> llext: found section symbol  addr 0x59c38[0m
[00:00:00.951,782] [0m<inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8[0m
[00:00:00.962,036] [0m<inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x59c24 addr 0x59c38[0m
*** Booting Zephyr OS build zephyr-v3.5.0-1327-g6bd0b54526cd ***
Running TESTSUITE llext
===================================================================
START - test_llext_simple
[00:00:00.408,660] [0m<dbg> llext: llext_load: Loading relocatable or shared elf[0m
[00:00:00.416,320] [0m<dbg> llext: do_llext_load: Finding ELF tables...[0m
[00:00:00.423,126] [0m<dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0[0m
[00:00:00.433,410] [0m<dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56[0m
[00:00:00.443,847] [0m<dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24[0m
[00:00:00.454,406] [0m<dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0[0m
[00:00:00.464,752] [0m<dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0[0m
[00:00:00.475,128] [0m<dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37[0m
[00:00:00.485,565] [0m<dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33[0m
[00:00:00.496,124] [0m<dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42[0m
[00:00:00.507,354] [0m<dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240[0m
[00:00:00.517,761] [0m<dbg> llext: llext_find_tables: symtab at 8[0m
[00:00:00.524,078] [0m<dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47[0m
[00:00:00.534,393] [0m<dbg> llext: llext_find_tables: strtab at 9[0m
[00:00:00.540,710] [0m<dbg> llext: llext_find_tables: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81[0m
[00:00:00.551,177] [0m<dbg> llext: llext_find_tables: shstrtab at 10[0m
[00:00:00.557,739] [0m<dbg> llext: do_llext_load: Allocate and copy strings...[0m
[00:00:00.565,124] [0m<dbg> llext: do_llext_load: Mapping ELF sections...[0m
[00:00:00.572,113] [0m<dbg> llext: llext_map_sections: section 0 name [0m
[00:00:00.578,796] [0m<dbg> llext: llext_map_sections: Not copied section [0m
[00:00:00.585,876] [0m<dbg> llext: llext_map_sections: section 1 name .text[0m
[00:00:00.593,017] [0m<dbg> llext: llext_map_sections: section 2 name .rel.text[0m
[00:00:00.600,494] [0m<dbg> llext: llext_map_sections: Not copied section .rel.text[0m
[00:00:00.608,337] [0m<dbg> llext: llext_map_sections: section 3 name .data[0m
[00:00:00.615,478] [0m<dbg> llext: llext_map_sections: section 4 name .bss[0m
[00:00:00.622,528] [0m<dbg> llext: llext_map_sections: section 5 name .rodata[0m
[00:00:00.629,852] [0m<dbg> llext: llext_map_sections: section 6 name .comment[0m
[00:00:00.637,237] [0m<dbg> llext: llext_map_sections: Not copied section .comment[0m
[00:00:00.644,989] [0m<dbg> llext: llext_map_sections: section 7 name .ARM.attributes[0m
[00:00:00.652,984] [0m<dbg> llext: llext_map_sections: Not copied section .ARM.attributes[0m
[00:00:00.661,346] [0m<dbg> llext: llext_map_sections: section 8 name .symtab[0m
[00:00:00.668,670] [0m<dbg> llext: llext_map_sections: Not copied section .symtab[0m
[00:00:00.676,330] [0m<dbg> llext: llext_map_sections: section 9 name .strtab[0m
[... 93 lines cut ...]
[00:00:00.758,209] [0m<dbg> llext: do_llext_load: Allocating memory for symbol table...[0m
[00:00:00.766,571] [0m<dbg> llext: do_llext_load: Copying symbols...[0m
[00:00:00.773,254] [0m<dbg> llext: llext_copy_symbols: function symbol 0 name hello_world addr 0x9b2d[0m
[00:00:00.782,775] [0m<dbg> llext: do_llext_load: Linking ELF...[0m
[00:00:00.789,123] [0m<dbg> llext: llext_link: relocation section .rel.text (2) linked to section 8 has 3 relocations[0m
[00:00:00.800,140] [0m<dbg> llext: llext_link: relocation 2:0 info 502 (type 2, sym 5) offset 40 sym_name  sym_type 3 sym_bind 0 sym_ndx 5[0m
[00:00:00.812,957] [0m<inf> llext: found section symbol  addr 0x9b68[0m
[00:00:00.819,641] [0m<inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8[0m
[00:00:00.830,108] [0m<inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x9b54 addr 0x9b68[0m
[00:00:00.839,660] [1;31m<err> os: ***** BUS FAULT *****[0m
[00:00:00.845,245] [1;31m<err> os:   Precise data bus error[0m
[00:00:00.851,104] [1;31m<err> os:   BFAR Address: 0x9b54[0m
[00:00:00.856,811] [1;31m<err> os: r0/a1:  0x20002270  r1/a2:  0x00009b54  r2/a3:  0x00009b68[0m
[00:00:00.865,661] [1;31m<err> os: r3/a4:  0x00000002 r12/ip:  0x000026cd r14/lr:  0x00003aa3[0m
[00:00:00.874,511] [1;31m<err> os:  xpsr:  0x61000000[0m
[00:00:00.879,852] [1;31m<err> os: Faulting instruction address (r15/pc): 0x00001cb0[0m
[00:00:00.887,939] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:00.895,935] [1;31m<err> os: Current thread: 0x200000f8 (test_llext_simple)[0m
[00:00:00.903,717] [1;31m<err> os: Halting system[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used: 26f30db8adabe40983ca2501e1c866dc7ebe2d5c

teburd, 2023-11-13:
@lyakh wonder if this is related to #64918

katgiadla, 2023-11-14:
PR #64918 does not resolve this problem.

teburd, 2023-11-14:
> PR #64918 does not resolve this problem.

The opposite is my suspicion, that this PR is the cause!

lyakh, 2023-11-14:
> PR #64918 does not resolve this problem.

@katgiadla sorry, could you confirm, that a snapshot before that PR passes the test and right after it fails?

katgiadla, 2023-11-15:
> > PR #64918 does not resolve this problem.
> 
> @katgiadla sorry, could you confirm, that a snapshot before that PR passes the test and right after it fails?

I've checked the tests results.
The last time, when these test configurations have passed, has been tested on version 1a0b1124c6294b9f3851b54fc7d0aeeba12a5f34.
Then PR has been merged with sha 0bf08e5775b4664cdef42a1c69785ab644a43462. Next test has executed on version 6bd0b54526cda9dcbcb77c0adf933517ca46fecd and since this test execution, described test configurations have failed.

lyakh, 2023-11-20:
@katgiadla could you please try the attached diff. I'm not sure which exactly ARM families are affected and whether this would be an acceptable work-around, but just to test my hypothesis for now.
[40.gitdiff.txt](https://github.com/zephyrproject-rtos/zephyr/files/13417167/40.gitdiff.txt)

teburd, 2023-11-21:
Well certainly in this case I think trying to write directly to flash as if it were writable memory might be an issue, the relocations are updating .text sections and with peek() this isn't possible.

The loader likely needs some logic to understand that if relocations (rewriting .text) is needed, and the elf is coming from what amounts to read-only memory (flash) then we need to copy .text over first.

lyakh, 2023-11-23:
> Well certainly in this case I think trying to write directly to flash as if it were writable memory might be an issue, the relocations are updating .text sections words and with peek() this isn't possible I imagine.

@teburd is it really about writing to flash? I thought this was similar to the failure that I observe with Cortex M3 in qemu. And there I think it's not about write-protected areas but about non-executable memory. But maybe this error is different, cannot test without the hardware, need  @katgiadla to test

FRASTM, 2023-11-24:
similar issue found on stm32 boards : https://github.com/zephyrproject-rtos/zephyr/issues/64581

closed 2023-12-13 by katgiadla