zephyrproject-rtos/zephyr #65138: STM32: support HSI48 as input for PLL

adolfogc, 2023-11-13:
Hi,

I'm interested in adding Zephyr support for a board based on a STM32F091xC MCU. This board comes with a STM32CubeMX project file detailing the following clock configuration:

[image]

Reading some of the applicable Zephyr source files, I came across this comment regarding the input source for the PLL:

https://github.com/zephyrproject-rtos/zephyr/blob/c9e6ac54d63875bfd463cf8f1eb3d3710de141a4/dts/bindings/clock/st%2Cstm32f0-pll-clock.yaml#L9C3-L10C50

What would it take to add support for this clock configuration?

erwango, 2023-11-14:
> What would it take to add support for this clock configuration?

You'll need to update this file:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/clock_control/clock_stm32f0_f3.c
Then add STM32_PLL_SRC_HSI48 in this file:
https://github.com/zephyrproject-rtos/zephyr/blob/4fe2605160fd535980924a7ea09080c5984fe896/include/zephyr/drivers/clock_control/stm32_clock_control.h#L221C1-L221C1
Then, add a dedicated test here:
tests/drivers/clock_control/stm32_clock_configuration/stm32_common_core

adolfogc, 2023-11-14:
Hi @erwango, thank you!