zephyrproject-rtos/zephyr #65144: net: loopback: add default IPv4 netmask of loopback interface for handling scenarios with multiple network interfaces

JungoLin1978, 2023-11-14:
**Is your enhancement proposal related to a problem? Please describe.**

In our product platform, we have two network interfaces: a mobile network interface for internet access and a loopback network interface for ADB forward function.
If both of these network interfaces are active, internet application data with IPv4 addresses are being routed to the loopback network instead of the mobile network interface as expected. Let's check the implementation of net_if_ipv4_select_src_iface at this link: https://elixir.bootlin.com/zephyr/v3.4.0/source/subsys/net/ip/net_if.c#L3182 to investigate this issue further. We believe that this behavior is caused by the netmask of the loopback interface since its IPv4 netmask is set to 0.0.0.0, resulting in it being chosen by the net_if_ipv4_select_src_iface function. However, when comparing it with Linux OS, we can see that in class A domain, the default IPv4 netmask for loopback is 255.0.0.0.

**Describe the solution you'd like**

According to the reference implementation of the Linux OS, we recommend adding a default IPv4 address for the loopback interface in the loopback_init function.
This addition will enhance routing in normal scenarios.

	if (IS_ENABLED(CONFIG_NET_IPV4)) {
		struct in_addr ipv4_loopback = INADDR_LOOPBACK_INIT;
                struct in_addr netmask = { { { 255, 0, 0, 0 } } };

		ifaddr = net_if_ipv4_addr_add(iface, &ipv4_loopback,
					      NET_ADDR_AUTOCONF, 0);
		if (!ifaddr) {
			LOG_ERR("Failed to register IPv4 loopback address");
		}
                net_if_ipv4_set_netmask(ipv4auto->iface, &netmask);                 
	}

rlubos, 2023-11-14:
It seems indeed that we should set the netmask as well when setting the default address on the loopback interface, in the place you point to. @JungoLin1978 Are you willing to send a PR with the change?

JungoLin1978, 2023-11-15:
Thank you for your feedback. I'm happy to submit a pull request with the changes, but I'm experiencing firewall issues that prevent me from creating a PR on my company's network. I will need to postpone this task and work on it over the weekend from my home.

rlubos, 2023-11-15:
@JungoLin1978 No worries, given it's such a simple change I've submitted PR already (https://github.com/zephyrproject-rtos/zephyr/pull/65210) so we can get this fixed right away.

JungoLin1978, 2023-11-15:
@rlubos  Thank you for your assistance, and I would like to close this issue.

PR #65210 (2023-11-15): net: loopback: Register IPv4 netmask
A proper netmask should be set on the loopback interface, so that source address selection work properly when there are multiple interfaces in the system.

Fixes #65144
closed 2023-11-15 by JungoLin1978