zephyrproject-rtos/zephyr #65153: driver: i3c: i3c_mcux: i3c tranfer timeout on rt595 platform

hakehuang, 2023-11-14:
**Describe the bug**
use i3c to access rt595 codec meet timeout and it is OK after revert below

1.     Revert "drivers: i3c: mcux: fix config_get to return last config set"
        This reverts commit ca3365864398b09935e63fa92c882b4a977dae8b.
2.     Revert "drivers: i3c: mcux: send 7h7e on first transfer or after stop"
        This reverts commit 86db0dc4874863988cea842a2e27ef2982ddfbca.
3.     Revert "drivers: i3c: mcux: tighten the FIFO read"
        This reverts commit faa722dbc501f88ce2ffd607b33942dcbfb0218d.
4.     Revert "drivers: i3c: mcux: Remove infinite wait for MCTRLDONE in auto-IBI"
        This reverts commit 4dc679068b2d1aaa3e239b13387ea739e7ac3309.
5.     Revert "drivers: i3c: mcux: Add dt property disable-open-drain-high-pp"
        This reverts commit 5f6502f67157f07bc0c110304e0b55923b545ded.

**To Reproduce**
Steps to reproduce the behavior:
clone code from 
https://github.com/hakehuang/zephyr.git br: main_br_rt595-enablement-dmic2
#west init -mr https://github.com/hakehuang/zephyr.git -br main_br_rt595-enablement-dmic2
#west update
# cd zephyr
# source ./zephyr-env.sh
in samples/driver/audio/dmic
1. west build -b mimxrt595_evk_cm33
2. west flash

if not revert above changes the log shows:

`
DMIC sample
audio codec ready
i2s ready
[00:09:31.563,212] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.564,491] <err> wolfson_wm8904: i2c write to codec error -116
--- 9 messages dropped ---
[00:09:31.565,769] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.567,048] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.568,326] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.569,605] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.570,883] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.572,161] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.573,440] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.574,719] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.575,998] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.577,277] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.578,555] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.579,834] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.582,387] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.584,942] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.587,495] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.590,050] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.592,601] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.595,153] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.597,722] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.598,999] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.601,551] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.602,835] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.606,668] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.609,222] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.611,778] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.614,334] <err> wolfson_wm8904: i2c write to codec error -116
[00:09:31.616,889] <err> wolfson_wm8904: i2c write to codec error -116
DMIC config OK
TX enabled
`

after revert above changes
`
DMIC sample
audio codec ready
i2s ready
[00:10:54.887,018] <inf> i2s_mcux_flexcomm: Device flexcomm@106000 inited
*** Booting Zephyr OS build v3.5.0-rc3-1187-gbc3bd27ea69b ***
CODEC WM8904 configreset 
DMIC config OK
TX enabled
========start loop============
`

and checking the waveform of i3s also shows with above commits the i3c does not work when driver calls it.

**Expected behavior**
I3C works

**Impact**
all drivers using i3c for mimxrtrt595_evk

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA or Version used: bc3bd27ea69b88f72fdfe5a8b44c66e51d052f19 (customerized)

decsny, 2024-06-14:
@hakehuang In case your codec PR doesn't merge to LTS you should fix the I3C bug in another PR

hakehuang, 2024-06-18:
> @hakehuang In case your codec PR doesn't merge to LTS you should fix the I3C bug in another PR

ok, let me create a PR

PR #74451 (2024-06-20): driver: i3c: fixing i3c init error
i3c config need has i2x/i3c baudrate configured first. so in i3c init we shall set the baudrate first

Fixes #65153
closed 2024-06-20 by aescolar