zephyrproject-rtos/zephyr #65170: modules: add heatshrink as external module

ClovisCorde, 2023-11-14:
## Origin

Name of project hosting the original open source code
[heatshrink: data compression/decompression library.](https://github.com/kickmaker/heatshrink)

## Purpose

Heatshrink is a data compression/decompression library for embedded/real-time systems.

## Mode of integration

As external module, as described in [heatshrink: zephyr-integration.](https://github.com/kickmaker/heatshrink#zephyr-integration-getting-started) 

## Maintainership

@RomainPelletant 
@ClovisCorde

## Pull Request

Pull request : none currently

## Description

[See the heatshrink API.](https://kickmaker.github.io/heatshrink/)

The primary functionality of heatshrink is to compress and decompress data.
Heatshrink is chosen because it has a small footprint memory (unlinke LZ4).

## Dependencies

No dependencies.

## Revision

bce543f477fa50cb1af5375f0c4271fd0ec83a93
(https://github.com/kickmaker/heatshrink contains modification for zephyr)

## License

Please use an SPDX identifier (https://spdx.org/licenses/), such as
``BSD-3-Clause``

ISC License

nordicjm, 2023-11-14:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/61822

closed 2023-11-14 by nordicjm

ClovisCorde, 2023-11-14:
Hi, the goal of [#61822 ](https://github.com/zephyrproject-rtos/zephyr/issues/61822) was to add heatshrink as an internal module, but we were told that it would be better if heatshrink was an [external module](https://github.com/zephyrproject-rtos/zephyr/issues/61822#issuecomment-1770459150) so this is the subject of this issue and the [PR #65171](https://github.com/zephyrproject-rtos/zephyr/pull/65171#event-10954388282)

nordicjm, 2023-11-14:
PR should be sufficient. There is no need for another issue when the original issue lists the outcome