zephyrproject-rtos/zephyr #65185: ntc-thermistor(-generic) needs an example

pem-dhoech, 2023-11-14:
Using the ntc-thermistor sensor driver is not straight forward. Setting it up similar to other drivers, which all use i2c, does not work.
An example would be really helpful.

kartben, 2023-11-14:
@asemjonovs would you be able to look into this? Would be nice to add an example of an overlay to the temp_polling code sample, and add a section to the README to indicate how the compensation table can be generated using the Python script?

asemjonovs, 2023-11-14:
@pem-dhoech what exactly is not working?    Not compiling, not getting any readings, bad readings, etc?

I found the thermometer sample to be really helpful in verifying the initial ntc thermistor changes.  I'd try compiling with that initially with your specific board and see what you get. See
https://docs.zephyrproject.org/latest/samples/sensor/thermometer/README.html
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/sensor/thermometer/

My initial changes have been refactored to match linux's implementation, plus a few additional changes after that, so I'm not sure if the compensation table generator still matches the latest.

kdunn926, 2023-12-13:
@asemjonovs - I also am having trouble using an NTC thermistor with the `samples/sensor/thermometer` example you shared.

Here is the overlay:
```
#include "epcos_thermistor.dtsi"

/ {
  mcp9700a: mcp9700a {
    status = "disabled";
    compatible = "microchip,mcp970x";
    family = "MCP9700/9700A";
    io-channels = <&adc 7>;
  };

  temp_sensor_ntc: temp_sensor_ntc {
    status = "okay";
    compatible = "zephyr,ntc-thermistor";
    io-channels = <&adc 7>;
    zephyr,rt-table = <&thermistor_R25_10000_B_3480>;
    r25-ohm = <10000>;
    pulldown-ohm = <10000>;
    connection-type = "NTC_CONNECTED_POSITIVE";
  };

  aliases {
    ambient-temp0 = &temp_sensor_ntc;
  };
};

&adc {
  #address-cells = <1>;
  #size-cells = <0>;

  channel@7 {
    reg = <7>;
    zephyr,gain = "ADC_GAIN_1_3";
    zephyr,reference = "ADC_REF_INTERNAL";
    zephyr,acquisition-time = <ADC_ACQ_TIME(ADC_ACQ_TIME_MICROSECONDS,10)>;
    zephyr,input-positive = <NRF_SAADC_AIN7>;
    zephyr,resolution = <12>; /* 0.055C per ADC step */
    zephyr,oversampling = <2>; /* x4 */
  };
};

```

Compiling fails with this error:
```
[ 92%] Linking C executable zephyr_pre0.elf
/opt/nordic/ncs/toolchains/4ef6631da0/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: ../app/libapp.a(main.c.obj): in function `main':
/opt/nordic/ncs/v2.4.1/zephyr/samples/sensor/thermometer/src/main.c:27: undefined reference to `__device_dts_ord_13'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/zephyr_pre0.elf] Error 1
make[1]: *** [zephyr/CMakeFiles/zephyr_pre0.dir/all] Error 2
make: *** [all] Error 2
```

asemjonovs, 2023-12-14:
Everything looks okay in your overlay from what I can see.  Seems like its this line in [main.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/sensor/thermometer/src/main.c#L57), right?
```
	const struct device *const dev = DEVICE_DT_GET(DT_ALIAS(ambient_temp0));
```
Take a look at Zephyr's [Devicetree Troubleshooting Guide](https://docs.zephyrproject.org/latest/build/dts/troubleshooting.html#undefined-reference-to-device-dts-ord-n) to see if anything there helps.  It should help guide you through your autogenerated files to hopefully identify your issue.

pem-dhoech, 2023-12-20:
Sorry for the late reply. After some digging around, looking at the sample from @kdunn926 and some experimenting, i got it working correctly. Here is how:
.overlay file:
````
/  {
	temp_sensor_ntc: temp_sensor_ntc {
		compatible = "ntc-thermistor-generic";
		status = "okay";
		io-channels = <&adc1 4>;
		pullup-uv = <3300000>;
		pullup-ohm = <10000>;
		pulldown-ohm = < 0 >;
		zephyr,compensation-table = 
		<(-40) 195652>,
		... "your compensation values" ...
		<(125) 531>;	
	};

	aliases {
		temp-housing = &temp_sensor_ntc;
	};
};

&adc1 {
	pinctrl-0 = <&adc1_in4_pc3>;
	pinctrl-names = "default";
	status = "okay";
	#address-cells = <1>;
	#size-cells = <0>;

	adc1_chan_4: channel@4 {	// AnalogIn A5, NTC Thermistor
		reg = <4>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
		zephyr,oversampling = <2>;
	};
};
````
main.cpp:
````
#include <zephyr/kernel.h>
#include <zephyr/drivers/sensor.h>

const struct device *const ntcCaseDev = DEVICE_DT_GET(DT_ALIAS(temp_housing));

struct sensor_value tempHousingValue;
double tempHousing;
const struct sensor_trigger tempHousingTrig = {
	.type = SENSOR_TRIG_THRESHOLD,
	.chan = SENSOR_CHAN_AMBIENT_TEMP,
};

int ret;
if (!device_is_ready(ntcCaseDev)) {
	printk("NTC device %s is not ready\n", ntcCaseDev->name);
	ret = 0;
}

ret = ReadTemperature(ntcCaseDev, &tempHousingValue);
tempHousing = sensor_value_to_double(&tempHousingValue);
````