zephyrproject-rtos/zephyr #65205: flexspi: ci: mimxrt101x serise CI broken due to flexRAM enabled

hakehuang, 2023-11-15:
**Describe the bug**
the mainline CI erro after download any application, the debuger can't acked anymore. have to use serial download mode to flash, and switch back to run. this will broken the CI.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt1015_evk/mimxrt1010_evk
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
git bisect found below commit cause this problem

```
commit 2d1fdb5586d219bb1e27e5e335bd216bf46a09c9
Author: Declan Snyder <declan.snyder@nxp.com>
Date:   Thu Sep 14 19:40:17 2023 -0500

    soc: nxp: rt1xxx: Enable NXP FlexRAM

    Enable NXP FlexRAM in DTS and SOC code.

    Do not configure flexram at runtime if the code is in the RAM.

    Fix RT1060 DT to be more accurate.

    Signed-off-by: Declan Snyder <declan.snyder@nxp.com>

 dts/arm/nxp/nxp_rt1010.dtsi                 |  9 ++++++
 dts/arm/nxp/nxp_rt1015.dtsi                 |  9 ++++++
 dts/arm/nxp/nxp_rt1020.dtsi                 | 13 +++++++++
 dts/arm/nxp/nxp_rt1024.dtsi                 | 13 +++++++++
 dts/arm/nxp/nxp_rt1040.dtsi                 |  4 +++
 dts/arm/nxp/nxp_rt1050.dtsi                 | 21 ++++++++++++++
 dts/arm/nxp/nxp_rt1060.dtsi                 | 44 +++++++++++++++++++++++------
 dts/arm/nxp/nxp_rt1064.dtsi                 | 21 ++++++++++++++
 dts/arm/nxp/nxp_rt10xx.dtsi                 |  7 ++++-
 dts/arm/nxp/nxp_rt11xx_cm7.dtsi             | 28 +++++++++++++++++-
 soc/arm/nxp_imx/rt/CMakeLists.txt           |  3 ++
 soc/arm/nxp_imx/rt/Kconfig.defconfig.series |  3 ++
 soc/arm/nxp_imx/rt/soc_rt10xx.c             |  7 +++++
 soc/arm/nxp_imx/rt/soc_rt11xx.c             |  8 +++++-

```

**To Reproduce**
Steps to reproduce the behavior:
1.cd samples/hello_workd
2. west build -b mimxrt1015_evk
3. west flash
4. and try west flash again
5. see error

**Expected behavior**
can continuous flash code

**Impact**
CI for rt101x series platform

**Logs and console output**
```
2023-11-15 14:47:25,586 - __main__ - INFO - ['C:\\Users\\nxa22633\\AppData\\Roaming\\Python\\Python39\\Scripts\\pyocd.exe', 'flash', '-u', '02260000070aec2f00000000000000000000000097969905', '-t', 'mimxrt1015', '-a', '0x60000000', 'Z:\\temp\\zephyr_rt1015.bin']
0002968 C No ACK received [__main__]
Traceback (most recent call last):
    value = result()
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 1060, in read_reg_cb
    res = transfer.get_result()
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 157, in get_result
    self.daplink.flush()
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\utility\concurrency.py", line 29, in _locking
    return func(self, *args, **kwargs)
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 870, in flush
    self._read_packet()
    return func(self, *args, **kwargs)
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 1146, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 480, in decode_data
    data = self._decode_transfer_data(data)
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 386, in _decode_transfer_data
    self._check_response(data[2])
  File "C:\Users\nxa22633\AppData\Roaming\Python\Python39\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 368, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used:  zephyr-v3.5.0-1512-g1200fce75c

hakehuang, 2023-11-15:
@decsny @dleach02 , please take a look, this blocks the CI for thoes two boards.

decsny, 2023-11-15:
I linked PR with fix, it's actually broken for all the RT10xx and RT11xx, the configuration was calculated wrong. I believe it only "broke" the RT101X because it only has 4 RAM banks, the ROM requires at least 2 banks to be configured as OCRAM, and so the likelihood of the wrong configuration having less than 2 OCRAM banks is higher on a part like RT101X which has less RAM banks.

PR #65256 (2023-11-15): drivers: nxp_flexram: Fix GPR 17 calculation
GPR17 calculation for configuration of RAM banks is incorrect, bit shift should be 2 per idx, not 1, this is major bug that needs correcting, currently all RT boards are affected with wrong configuration.

Fixes #65205
closed 2023-11-15 by mmahadevan108