zephyrproject-rtos/zephyr #65208: tests-ci : llext: simple: arm test Timeout

hakehuang, 2023-11-15:
**Describe the bug**
arm test is Timeout on zephyr-v3.5.0-1427-gc1a6cfdf9bd8 on lpcxpresso54114_m4
testcase path is llext/simple/arm/llext_simple

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso54114_m4  --sub-test llext.simple
```
or
```
# cd tests/llext/simple/arm/llext_simple
# west build -b lpcxpresso54114_m4
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.5.0-1427-gc1a6cfdf9bd8 ***
Running TESTSUITE llext
===================================================================
START - test_llext_simple
[00:00:00.016,000] [0m<dbg> llext: llext_load: Loading relocatable or shared elf[0m
[00:00:00.023,000] [0m<dbg> llext: do_llext_load: Finding ELF tables...[0m
[00:00:00.030,000] [0m<dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0[0m
[00:00:00.040,000] [0m<dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56[0m
[00:00:00.051,000] [0m<dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24[0m
[00:00:00.061,000] [0m<dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0[0m
[00:00:00.072,000] [0m<dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0[0m
[00:00:00.082,000] [0m<dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37[0m
[00:00:00.092,000] [0m<dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33[0m
[00:00:00.103,000] [0m<dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42[0m
[00:00:00.114,000] [0m<dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240[0m
[00:00:00.124,000] [0m<dbg> llext: llext_find_tables: symtab at 8[0m
[00:00:00.131,000] [0m<dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47[0m
[00:00:00.141,000] [0m<dbg> llext: llext_find_tables: strtab at 9[0m
[00:00:00.147,000] [0m<dbg> llext: llext_find_tables: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81[0m
[00:00:00.158,000] [0m<dbg> llext: llext_find_tables: shstrtab at 10[0m
[00:00:00.164,000] [0m<dbg> llext: do_llext_load: Allocate and copy strings...[0m
[00:00:00.171,000] [0m<dbg> llext: do_llext_load: Mapping ELF sections...[0m
[00:00:00.178,000] [0m<dbg> llext: llext_map_sections: section 0 name [0m
[00:00:00.185,000] [0m<dbg> llext: llext_map_sections: Not copied section [0m
[00:00:00.192,000] [0m<dbg> llext: llext_map_sections: section 1 name .text[0m
[00:00:00.199,000] [0m<dbg> llext: llext_map_sections: section 2 name .rel.text[0m
[00:00:00.207,000] [0m<dbg> llext: llext_map_sections: Not copied section .rel.text[0m
[00:00:00.214,000] [0m<dbg> llext: llext_map_sections: section 3 name .data[0m
[00:00:00.222,000] [0m<dbg> llext: llext_map_sections: section 4 name .bss[0m
[00:00:00.229,000] [0m<dbg> llext: llext_map_sections: section 5 name .rodata[0m
[00:00:00.236,000] [0m<dbg> llext: llext_map_sections: section 6 name .comment[0m
[00:00:00.243,000] [0m<dbg> llext: llext_map_sections: Not copied section .comment[0m
[00:00:00.251,000] [0m<dbg> llext: llext_map_sections: section 7 name .ARM.attributes[0m
[00:00:00.259,000] [0m<dbg> llext: llext_map_sections: Not copied section .ARM.attributes[0m
[00:00:00.267,000] [0m<dbg> llext: llext_map_sections: section 8 name .symtab[0m
[00:00:00.274,000] [0m<dbg> llext: llext_map_sections: Not copied section .symtab[0m
[00:00:00.282,000] [0m<dbg> llext: llext_map_sections: section 9 name .strtab[0m
[00:00:00.289,000] [0m<dbg> llext: llext_map_sections: Not copied section .strtab[0m
[00:00:00.297,000] [0m<dbg> llext: llext_map_sections: section 10 name .shstrtab[0m
[00:00:00.305,000] [0m<dbg> llext: llext_map_sections: Not copied section .shstrtab[0m
[00:00:00.312,000] [0m<dbg> llext: do_llext_load: Allocate and copy sections...[0m
[00:00:00.320,000] [0m<dbg> llext: do_llext_load: Counting exported symbols...[0m
[00:00:00.327,000] [0m<dbg> llext: llext_count_export_syms: symbol count 15[0m
[00:00:00.334,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 0, name , type tag 0, bind 0, sect 0[0m
[00:00:00.345,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 1, name hello_world.c, type tag 4, bind 0, sect 65521[0m
[00:00:00.357,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 2, name , type tag 3, bind 0, sect 1[0m
[00:00:00.367,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 3, name , type tag 3, bind 0, sect 3[0m
[00:00:00.377,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 4, name , type tag 3, bind 0, sect 4[0m
[00:00:00.388,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 5, name , type tag 3, bind 0, sect 5[0m
[00:00:00.398,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 6, name $d, type tag 0, bind 0, sect 5[0m
[00:00:00.409,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 7, name number, type tag 1, bind 0, sect 5[0m
[00:00:00.420,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 8, name $t, type tag 0, bind 0, sect 1[0m
[00:00:00.430,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 9, name $d, type tag 0, bind 0, sect 1[0m
[00:00:00.441,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 10, name $t, type tag 0, bind 0, sect 1[0m
[00:00:00.451,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 11, name , type tag 3, bind 0, sect 6[0m
[00:00:00.462,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 12, name , type tag 3, bind 0, sect 7[0m
[00:00:00.472,000] [0m<dbg> llext: llext_count_export_syms: function symbol 13, name hello_world, type tag 2, bind 1, sect 1[0m
[00:00:00.484,000] [0m<dbg> llext: llext_count_export_syms: unhandled symbol 14, name printk, type tag 0, bind 1, sect 0[0m
[00:00:00.495,000] [0m<dbg> llext: do_llext_load: Allocating memory for symbol table...[0m
[00:00:00.503,000] [0m<dbg> llext: do_llext_load: Copying symbols...[0m
[00:00:00.509,000] [0m<dbg> llext: llext_copy_symbols: function symbol 0 name hello_world addr 0x7a5d[0m
[00:00:00.519,000] [0m<dbg> llext: do_llext_load: Linking ELF...[0m
[00:00:00.525,000] [0m<dbg> llext: llext_link: relocation section .rel.text (2) linked to section 8 has 3 relocations[0m
[00:00:00.536,000] [0m<dbg> llext: llext_link: relocation 2:0 info 502 (type 2, sym 5) offset 40 sym_name  sym_type 3 sym_bind 0 sym_ndx 5[0m
[00:00:00.548,000] [0m<inf> llext: found section symbol  addr 0x7a98[0m
[00:00:00.555,000] [0m<inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8[0m
[00:00:00.565,000] [0m<inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x7a84 addr 0x7a98[0m
[00:00:00.574,000] [1;31m<err> os: ***** BUS FAULT *****[0m
[00:00:00.580,000] [1;31m<err> os:   Imprecise data bus error[0m
[00:00:00.586,000] [1;31m<err> os: r0/a1:  0x200020d8  r1/a2:  0x00007a84  r2/a3:  0x00007a98[0m
[00:00:00.594,000] [1;31m<err> os: r3/a4:  0x00000002 r12/ip:  0x00001e4d r14/lr:  0x0000331b[0m
[00:00:00.603,000] [1;31m<err> os:  xpsr:  0x61000000[0m
[00:00:00.608,000] [1;31m<err> os: Faulting instruction address (r15/pc): 0x0000331a[0m
[00:00:00.616,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:00.624,000] [1;31m<err> os: Current thread: 0x20000058 (test_llext_simple)[0m
[00:00:00.631,000] [1;31m<err> os: Halting system[0m

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.5.0-1427-gc1a6cfdf9bd8

hakehuang, 2023-11-15:
@dleach02  looks like memory size issue. test overflowed

nashif, 2023-11-21:
@teburd

teburd, 2023-11-21:
The test works ok for me on mimxrt1010 so I'm surprised to see that it overflows on a much larger device

```
$ west build -p always -b mimxrt1010_evk tests/subsys/llext -- -DCONFIG_ARM_MPU=n
-- west build: making build dir /home/tburdick/z/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/tburdick/z/zephyr/tests/subsys/llext
-- CMake version: 3.25.3
-- Found Python3: /home/tburdick/z/env/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/tburdick/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/tburdick/z/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1010_evk
-- Found host-tools: zephyr 0.16.0 (/nix/store/y033c1zj8h064wm8kbnpmfcf6w54cn3z-zephyr-sdk)
-- Found toolchain: zephyr 0.16.0 (/nix/store/y033c1zj8h064wm8kbnpmfcf6w54cn3z-zephyr-sdk)
-- Found Dtc: /nix/store/y033c1zj8h064wm8kbnpmfcf6w54cn3z-zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/tburdick/z/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk.dts
-- Generated zephyr.dts: /home/tburdick/z/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/tburdick/z/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/tburdick/z/zephyr/build/zephyr/dts.cmake

warning: HW_STACK_PROTECTION (defined at arch/Kconfig:259) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: ARCH_HAS_STACK_PROTECTION (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_HW_STACK_PROTECTION and/or look up
HW_STACK_PROTECTION in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /home/tburdick/z/zephyr/Kconfig
Loaded configuration '/home/tburdick/z/zephyr/boards/arm/mimxrt1010_evk/mimxrt1010_evk_defconfig'
Merged configuration '/home/tburdick/z/zephyr/tests/subsys/llext/prj.conf'
Merged configuration '/home/tburdick/z/zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/home/tburdick/z/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/tburdick/z/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /nix/store/y033c1zj8h064wm8kbnpmfcf6w54cn3z-zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/y033c1zj8h064wm8kbnpmfcf6w54cn3z-zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
Load components for MIMXRT1011:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpuart component is included.
driver_igpio component is included.
driver_cache_armv7_m7 component is included.
Loading Zephyr default modules (Zephyr base (cached)).
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tburdick/z/zephyr/build
-- west build: building application
[1/143] Preparing syscall dependency handling

[3/143] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/tburdick/z/zephyr), build: zephyr-v3.5.0-1662-gfd129de62879
[143/143] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       50104 B        16 MB      0.30%
             RAM:       45112 B        64 KB     68.84%
            ITCM:          0 GB        32 KB      0.00%
            DTCM:          0 GB        32 KB      0.00%
           OCRAM:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%

$ west flash -r pyocd
-- west flash: using runner pyocd
WARNING: runners.pyocd: hex file (None) does not exist; falling back on .bin (/home/tburdick/z/zephyr/build/zephyr/zephyr.bin). Consider enabling CONFIG_BUILD_OUTPUT_HEX.
[... 76 lines cut ...]
[00:00:00.580,000] <inf> llext: found symbol printk at 0x60008575, updating op code 0x6000964c
[00:00:00.589,000] <inf> llext: relocating (linking) symbol printk type 0 binding 1 ndx 0 offset 44 link section 8
[00:00:00.600,000] <inf> llext: writing relocation symbol printk type 2 sym 14 at addr 0x6000964c addr 0x60008575
[00:00:00.610,000] <dbg> llext: llext_link: relocation 2:2 info 502 (type 2, sym 5) offset 48 sym_name  sym_type 3 sym_bind 05
[00:00:00.623,000] <inf> llext: found section symbol  addr 0x6000966c
[00:00:00.630,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 48 link section 8
[00:00:00.640,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x60009650 addr 0x6000966c
[00:00:00.650,000] <dbg> llext: do_llext_load: loaded module, .text at 0x60009620, .rodata at 0x60009658
[00:00:00.660,000] <inf> llext: Loaded extension hello
hello world
A number is 42
 PASS - test_llext_simple in 0.653 seconds
===================================================================
TESTSUITE llext succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [llext]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.653 seconds
 - PASS - [llext.test_llext_simple] duration = 0.653 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

teburd, 2023-11-21:
I tried on my 1060evk as well and can confirm it works with the latest main. I don't have any lpc boards to try this on.

hakehuang, 2023-11-22:
@teburd rt1060 is OK in latest main, but lpcxpresso54114_m4 still fails.

```
[00:00:00.207,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .rel.text[0m
[00:00:00.214,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 3 name .data[0m
[00:00:00.222,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 4 name .bss[0m
[00:00:00.229,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 5 name .rodata[0m
[00:00:00.236,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 6 name .comment[0m
[00:00:00.243,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .comment[0m
[00:00:00.251,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 7 name .ARM.attributes[0m
[00:00:00.259,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .ARM.attributes[0m
[00:00:00.267,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 8 name .symtab[0m
[00:00:00.274,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .symtab[0m
[00:00:00.282,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 9 name .strtab[0m
[00:00:00.289,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .strtab[0m
[00:00:00.297,000] [0m&lt;dbg&gt; llext: llext_map_sections: section 10 name .shstrtab[0m
[00:00:00.305,000] [0m&lt;dbg&gt; llext: llext_map_sections: Not copied section .shstrtab[0m
[00:00:00.312,000] [0m&lt;dbg&gt; llext: do_llext_load: Allocate and copy sections...[0m
[00:00:00.320,000] [0m&lt;dbg&gt; llext: do_llext_load: Counting exported symbols...[0m
[00:00:00.327,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: symbol count 15[0m
[00:00:00.334,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 1, name hello_world.c, type tag 4, bind 0, sect 65521[0m
[00:00:00.346,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 2, name , type tag 3, bind 0, sect 1[0m
[00:00:00.357,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 3, name , type tag 3, bind 0, sect 3[0m
[00:00:00.367,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 4, name , type tag 3, bind 0, sect 4[0m
[00:00:00.377,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 5, name , type tag 3, bind 0, sect 5[0m
[00:00:00.388,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 6, name $d, type tag 0, bind 0, sect 5[0m
[00:00:00.398,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 7, name number, type tag 1, bind 0, sect 5[0m
[00:00:00.409,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 8, name $t, type tag 0, bind 0, sect 1[0m
[00:00:00.420,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 9, name $d, type tag 0, bind 0, sect 1[0m
[00:00:00.430,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 10, name $t, type tag 0, bind 0, sect 1[0m
[00:00:00.441,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 11, name , type tag 3, bind 0, sect 6[0m
[00:00:00.452,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 12, name , type tag 3, bind 0, sect 7[0m
[00:00:00.462,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: function symbol 13, name hello_world, type tag 2, bind 1, sect 1[0m
[00:00:00.473,000] [0m&lt;dbg&gt; llext: llext_count_export_syms: unhandled symbol 14, name printk, type tag 0, bind 1, sect 0[0m
[00:00:00.484,000] [0m&lt;dbg&gt; llext: do_llext_load: Allocating memory for symbol table...[0m
[00:00:00.493,000] [0m&lt;dbg&gt; llext: do_llext_load: Copying symbols...[0m
[00:00:00.499,000] [0m&lt;dbg&gt; llext: llext_copy_symbols: function symbol 0 name hello_world addr 0x8e5d[0m
[00:00:00.508,000] [0m&lt;dbg&gt; llext: do_llext_load: Linking ELF...[0m
[00:00:00.515,000] [0m&lt;dbg&gt; llext: llext_link: relocation section .rel.text (2) linked to section 8 has 3 relocations[0m
[00:00:00.525,000] [0m&lt;dbg&gt; llext: llext_link: relocation 2:0 info 502 (type 2, sym 5) offset 40 sym_name  sym_type 3 sym_bind 0 sym_ndx 5[0m
[00:00:00.538,000] [0m&lt;inf&gt; llext: found section symbol  addr 0x8e98[0m
[00:00:00.544,000] [0m&lt;inf&gt; llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8[0m
[00:00:00.555,000] [0m&lt;inf&gt; llext: writing relocation symbol  type 2 sym 5 at addr 0x8e84 addr 0x8e98[0m
[00:00:00.564,000] [1;31m&lt;err&gt; os: ***** BUS FAULT *****[0m
[00:00:00.569,000] [1;31m&lt;err&gt; os:   Imprecise data bus error[0m
[00:00:00.575,000] [1;31m&lt;err&gt; os: r0/a1:  0x200020e0  r1/a2:  0x00008e84  r2/a3:  0x00008e98[0m
[00:00:00.584,000] [1;31m&lt;err&gt; os: r3/a4:  0x00000002 r12/ip:  0x00000000 r14/lr:  0x0000339b[0m
[00:00:00.593,000] [1;31m&lt;err&gt; os:  xpsr:  0x61000000[0m
[00:00:00.598,000] [1;31m&lt;err&gt; os: Faulting instruction address (r15/pc): 0x0000339a[0m
[00:00:00.606,000] [1;31m&lt;err&gt; os: &gt;&gt;&gt; ZEPHYR FATAL ERROR 0: CPU exception on CPU 0[0m
[00:00:00.614,000] [1;31m&lt;err&gt; os: Current thread: 0x20000058 (test_llext_simple)[0m
[00:00:00.621,000] [1;31m&lt;err&gt; os: Halting system[0m

```

teburd, 2023-11-22:
@hakehuang can you try to bisect? There’s some changes that might’ve cause this.

hakehuang, 2023-11-23:
> @hakehuang can you try to bisect? There’s some changes that might’ve cause this.

this one 

'''
0bf08e5775b4664cdef42a1c69785ab644a43462 is the first bad commit
commit 0bf08e5775b4664cdef42a1c69785ab644a43462
Author: Guennadi Liakhovetski <guennadi.liakhovetski@linux.intel.com>
Date:   Wed Oct 25 17:05:38 2023 +0200

    llext: use llext_peek() for section pointers

    Try to use llext_peek() for section pointers. If it's supported and
    succeeds we don't need to allocate buffers.

    Signed-off-by: Guennadi Liakhovetski <guennadi.liakhovetski@linux.intel.com>

:040000 040000 aff80e8b0c70069b10a7179859607050b8cdf6c7 c73fdb156706ca6b0386a2736de2362b8acd5189 M      include
:040000 040000 d2bb70bfa2f935f5cd8419c593b48c3ae9a47185 657aea63652a7acaa598525d1db166a946b8a5e7 M      subsys

'''

teburd, 2024-01-12:
I don't have a lpc part to test on or time to dig into why this is an issue, reassigning to the offending commit author for now.

hakehuang, 2024-01-14:
> I don't have a lpc part to test on or time to dig into why this is an issue, reassigning to the offending commit author for now. There have been fixes to llext since this was created and it would be interesting to see if this is still an issue @hakehuang

pre latest testing zephyr-v3.5.0-3466-gb702c8e8f01f, this issue is fiex for lpc54114_m4 platform.

closed 2024-01-14 by hakehuang