zephyrproject-rtos/zephyr #65220: subsys: logging: log_core_additional: logging.sync and logging.async fails

katgiadla, 2023-11-15:
**Describe the bug**
The tests `logging.sync` and `logging.async` fail.

Observed for:
- `nrf52dk_nrf52832`
- `nrf52840dk_nrf52840`
- `nrf5340dk_nrf5340_cpuapp`
- `nrf5340dk_nrf5340_cpuapp_ns`
- `nrf9160dk_nrf9160`
- `nrf9160dk_nrf9160_ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk_nrf52840 connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/subsys/logging/log_core_additional -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
For `logging.sync`: 

```
*** Booting Zephyr OS build zephyr-v3.5.0-1427-gc1a6cfdf9bd8 ***
Running TESTSUITE test_log_core_additional
===================================================================
START - test_log_backend
PASS - test_log_backend in 0.001 seconds
===================================================================
START - test_log_domain_id
[00:00:00.557,739] [0m<inf> log_test: info message for domain id test[0m
PASS - test_log_domain_id in 0.007 seconds
===================================================================
START - test_log_early_logging
SKIP - test_log_early_logging in 0.001 seconds
===================================================================
START - test_log_generic
[00:00:00.589,813] [0m<inf> log generic[0m
[00:00:00.593,902] [0m<inf> log generic: log user space[0m
[00:00:00.599,365] [0m<inf> log generic 100
[0m
[00:00:00.603,912] [0m<inf> log generic 100, 1
[0m
PASS - test_log_generic in 0.019 seconds
===================================================================
START - test_log_msg_create
PASS - test_log_msg_create in 0.001 seconds
===================================================================
START - test_log_severity
[00:00:00.633,453] [0m<inf> log_test: info message[0m
[00:00:00.638,519] [1;33m<wrn> log_test: warning message[0m
[00:00:00.644,104] [1;31m<err> log_test: error message[0m
PASS - test_log_severity in 0.017 seconds
===================================================================
START - test_log_sync
Logging synchronously
[00:00:00.663,391] [0m<inf> log_test: Log immediately[0m
[00:00:00.668,701] [0m<inf> log_test: Log immediately[0m
PASS - test_log_sync in 0.013 seconds
===================================================================
START - test_log_thread
SKIP - test_log_thread in 0.001 seconds
===================================================================
START - test_log_timestamping
Register timestamp function
[00:00:00.000,000] [0m<inf> log_test: test timestamp[0m
[00:00:00.000,001] [0m<inf> log_test: test timestamp[0m
[00:00:00.000,001] [1;33m<wrn> log_test: test timestamp[0m
PASS - test_log_timestamping in 0.019 seconds
===================================================================
START - test_multiple_backends
Test multiple backends
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/logging/log_core_additional/src/log_test.c:399: test_log_core_additional_test_multiple_backends: (have_uart is false)
There is no UART log backend found
FAIL - test_multiple_backends in 0.023 seconds
===================================================================
TESTSUITE test_log_core_additional failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  87.50% [test_log_core_additional]: pass = 7, fail = 1, skip = 2, total = 10 duration = 0.102 seconds
- PASS - [test_log_core_additional.test_log_backend] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_domain_id] duration = 0.007 seconds
- SKIP - [test_log_core_additional.test_log_early_logging] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_generic] duration = 0.019 seconds
- PASS - [test_log_core_additional.test_log_msg_create] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_severity] duration = 0.017 seconds
- PASS - [test_log_core_additional.test_log_sync] duration = 0.013 seconds
- SKIP - [test_log_core_additional.test_log_thread] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_timestamping] duration = 0.019 seconds
- FAIL - [test_log_core_additional.test_multiple_backends] duration = 0.023 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 0bbbb1189a0d7912718bfe8e4af8d692
PROJECT EXECUTION FAILED
```

For `logging.async`:
*** Booting Zephyr OS build zephyr-v3.5.0-1427-gc1a6cfdf9bd8 ***
Running TESTSUITE test_log_core_additional
===================================================================
START - test_log_backend
PASS - test_log_backend in 0.001 seconds
===================================================================
START - test_log_domain_id
[00:00:00.565,185] [0m<inf> log_test: info message for domain id test[0m
PASS - test_log_domain_id in 0.008 seconds
===================================================================
START - test_log_early_logging
Create log message before backend active
Activate backend with context PASS - test_log_early_logging in 0.008 seconds
===================================================================
START - test_log_generic
PASS - test_log_generic in 0.002 seconds
===================================================================
START - test_log_msg_create
PASS - test_log_msg_create in 0.002 seconds
===================================================================
START - test_log_severity
PASS - test_log_severity in 0.002 seconds
===================================================================
START - test_log_sync
Logging synchronously
SKIP - test_log_sync in 0.003 seconds
===================================================================
START - test_log_thread
SKIP - test_log_thread in 0.001 seconds
===================================================================
START - test_log_timestamping
Register timestamp function
PASS - test_log_timestamping in 0.004 seconds
===================================================================
START - test_multiple_backends
Test multiple backends
Assertion failed at ../../../../../../../../../zephyr/zephyr/tests/subsys/logging/log_core_additional/src/log_test.c:399: test_log_core_additional_test_multiple_backends: (have_uart is false)
There is no UART log backend found
FAIL - test_multiple_backends in 0.023 seconds
===================================================================
TESTSUITE test_log_core_additional failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  87.50% [test_log_core_additional]: pass = 7, fail = 1, skip = 2, total = 10 duration = 0.054 seconds
- PASS - [test_log_core_additional.test_log_backend] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_domain_id] duration = 0.008 seconds
- PASS - [test_log_core_additional.test_log_early_logging] duration = 0.008 seconds
- PASS - [test_log_core_additional.test_log_generic] duration = 0.002 seconds
- PASS - [test_log_core_additional.test_log_msg_create] duration = 0.002 seconds
- PASS - [test_log_core_additional.test_log_severity] duration = 0.002 seconds
- SKIP - [test_log_core_additional.test_log_sync] duration = 0.003 seconds
- SKIP - [test_log_core_additional.test_log_thread] duration = 0.001 seconds
- PASS - [test_log_core_additional.test_log_timestamping] duration = 0.004 seconds
- FAIL - [test_log_core_additional.test_multiple_backends] duration = 0.023 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 862895f18e7be57ea056d69f74495271
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used: zephyr-v3.5.0-1427-gc1a6cfdf9bd8

nordic-krch, 2023-11-15:
Fixed by #65150

nordic-krch, 2023-11-15:
Duplicate of #65151.

closed 2023-11-15 by nordic-krch