zephyrproject-rtos/zephyr #65267: Zynq SDHCI driver support

panxxhub, 2023-11-16:
**Is your feature request related to a problem? Please describe.**
Zynq's SD Controller is not supported yet.

**Describe the solution you'd like**
Add a device driver, which would be 'xilinx,sdhci-8.9a' compatible.

**Describe alternatives you've considered**
since many soc share the same sd controller ip which is 'arasan,sdhci', shall we add this vendor to binding index.

henrikbrixandersen, 2023-11-16:
Hi! That's a good idea, and a nice a addition. Feel free to submit a PR for implementing this.

closed 2023-11-16 by henrikbrixandersen

m31nr4d, 2026-09-01:
I have a working, hardware-verified Arasan SDHCI implementation for Zynq-7000 (Trenz TE0726, xc7z010) -- PIO/polling, register-level, tested against a real microSD card (init/read/write/verify all green). It's currently written as an ad-hoc disk_operations-backed driver in a private out-of-tree project, not yet in Zephyr's driver-model shape (no DT binding, hardcoded base address, no DEVICE_DT_INST_DEFINE).

Before I invest the time to port it to a proper devicetree-bound driver under drivers/disk/, I wanted to check: is this still wanted, and does anyone have opinions on scope (e.g. keep v1 PIO-only vs. expect ADMA2/interrupt-driven from the start, arasan,sdhci vs. a Xilinx-specific compatible)? Happy to open a draft PR once I have a driver-model version.