zephyrproject-rtos/zephyr #65280: STM32H503 - simple board fails compiling

Sam3e2, 2023-11-16:
**Describe the bug**

Creating a simple board using a STM32H503 fails. Tested with main line and v.3.5.0 release. The build fails, because there are unknown types in the HAL driver:
- unknown type name 'DCACHE_TypeDef'
- unknown type name 'DLYB_TypeDef'
- unknown type name 'SAI_Block_TypeDef'

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to samples/basic/minimal
2. Extract [boards.zip](https://github.com/zephyrproject-rtos/zephyr/files/13375378/boards.zip)
3. west build -b h5test -p -- -DBOARD_ROOT=.

The h563test board is working.

**Expected behavior**
A successful build

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - Version: v3.5.0

FRASTM, 2023-11-16:
Hello @Sam3e2, 
I see that pb of such a new stm32h503 board 
...investigating

closed 2023-11-22 by carlescufi

Sam3e2, 2024-02-23:
It fails again on the main branch:

```
zephyrproject\zephyr\modules\cmsis\cmsis_core_m.h:68:2: error: #error "__SAUREGION_PRESENT and CONFIG_CPU_HAS_ARM_SAU are not set to the same value"
   68 | #error "__SAUREGION_PRESENT and CONFIG_CPU_HAS_ARM_SAU are not set to the same value"
```

The h563test board builds still without any problems. It seems that CONFIG_CPU_HAS_ARM_SAU isn't set properly. __SAUREGION_PRESENT is set in a cmsis header and therefore mismatches CONFIG_CPU_HAS_ARM_SAU.

erwango, 2024-07-02:
> It fails again on the main branch:

@Sam3e2 People don't pay attention to close tickets. Next time please open a new ticket.
Regarding the issue, it seems there is a disconnection between Ref Man that doesn't mention SAU presence on H503 and CMSIS file which states, in `hal_stm32/stm32cube/stm32h5xx/soc/stm32h503xx.h`:
[`#define __SAUREGION_PRESENT       1U        /* SAU regions present */`](https://github.com/zephyrproject-rtos/hal_stm32/blob/855f195793df094644ce8b8617c01275408bbf58/stm32cube/stm32h5xx/soc/stm32h503xx.h#L178-L179)

In zephyr code, we've followed Ref man recommendation, using the following code, with specify SAU is not available on H503
https://github.com/zephyrproject-rtos/zephyr/blob/c10e3d581760f814c0be679a4e7bf9b104541395/soc/st/stm32/stm32h5x/Kconfig#L9-L10

According to ref man, the correct fix is to state SAU is not present in CMSIS file:
```diff
-#define __SAUREGION_PRESENT       1U        /* SAU regions present */
+#define __SAUREGION_PRESENT       0U        /* SAU regions not present */
```