zephyrproject-rtos/zephyr #65322: net: tls_credentials cmake is wrong, causing build failure

decsny, 2023-11-16:
**Describe the bug**
subsys/net/lib/tls_credentials folder is not building correctly because of missing link interface connections, see build output below of tests/net/lib/tls_credentials on an RT board

The issue is that the mbedTLS interface is not being linked to these sources.

**To Reproduce**
Steps to reproduce the behavior:
`west build -b mimxrt1060_evk -p always tests/net/lib/tls_credentials/`

**Expected behavior**
source files build

**Impact**
Blocking #62833

**Logs and console output**
```
-- west build: building application
[1/279] Preparing syscall dependency handling

[2/279] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/declan/zephyrproject/zephyr), build: zephyr-v3.5.0-1661-g256adeebd934
[58/279] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c.obj 
ccache /home/declan/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCPU_MIMXRT1062DVL6A -DKERNEL -DPICOLIBC_INTEGER_PRINTF_SCANF -DXIP_BOOT_HEADER_DCD_ENABLE=1 -DXIP_BOOT_HEADER_ENABLE=1 -DXIP_EXTERNAL_FLASH -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/declan/zephyrproject/zephyr/kernel/include -I/home/declan/zephyrproject/zephyr/arch/arm/include -I/home/declan/zephyrproject/zephyr/drivers/memc -I/home/declan/zephyrproject/zephyr/include -I/home/declan/zephyrproject/zephyr/build/zephyr/include/generated -I/home/declan/zephyrproject/zephyr/soc/arm/nxp_imx/rt -I/home/declan/zephyrproject/zephyr/subsys/testsuite/include -I/home/declan/zephyrproject/zephyr/subsys/testsuite/ztest/include -I/home/declan/zephyrproject/zephyr/subsys/net/l2 -I/home/declan/zephyrproject/zephyr/subsys/net/lib/tls_credentials/. -I/home/declan/zephyrproject/zephyr/subsys/net/lib/dns/. -I/home/declan/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/declan/zephyrproject/zephyr/modules/cmsis/. -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062 -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062/drivers -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1062/xip -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/trng -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/enet -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7 -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyksz8081 -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyar8031 -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phylan8720a -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyrtl8211f -I/home/declan/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/components/phy/mdio/enet -I/home/declan/zephyrproject/zephyr/modules/hal_nxp/. -isystem /home/declan/zephyrproject/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros /home/declan/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/declan/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/declan/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/declan/zephyrproject/zephyr/tests/net/lib/tls_credentials=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/declan/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/declan/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/CMakeFiles/zephyr.dir/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c.obj -MF zephyr/CMakeFiles/zephyr.dir/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c.obj -c /home/declan/zephyrproject/zephyr/subsys/net/lib/tls_credentials/tls_credentials_digest_raw.c
In file included from <command-line>:
/home/declan/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h:217:33: fatal error: config-tls-generic.h: No such file or directory
  217 | #define CONFIG_MBEDTLS_CFG_FILE "config-tls-generic.h"
      |                                 ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[67/279] Building C object zephyr/arch/arch/arm/core/cortex_m/CMakeFiles/arch__arm__core__cortex_m.dir/irq_init.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/declan/zephyrproject/zephyr/build
```

bperseghetti, 2023-11-21:
Still having issues after rebase: https://github.com/zephyrproject-rtos/zephyr/actions/runs/6938290433

Should have been fixed with merge of https://github.com/zephyrproject-rtos/zephyr/pull/65323 but appears to still be a problem after rerunning tests.

bperseghetti, 2023-11-21:
Wondering if it might be CI cache? It does not fail on local machine anymore.

rlubos, 2023-11-21:
If you check the revision, used for the linked run:
[image]
It does not seem to have fixes included in the commit history: https://github.com/zephyrproject-rtos/zephyr/commits/b6c60f034b404d83b046ee77910a078c89a2bc18. So likely, yes, that run still used some older Zephyr commit as a base.

PR #65323 (2023-11-21): net: Fix CMakeLists
Fix the CMakeLists of the tls_credentials and sockets folder to link/interface to the net library instead of the zephyr library. This fixes issues where some files are not found in the link interface when compiling the sources in this folder.

Fixes #65322
closed 2023-11-21 by decsny