zephyrproject-rtos/zephyr #65347: [Coverity CID: 329999] Untrusted loop bound in subsys/net/ip/dhcpv6.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/net/ip/dhcpv6.c

Category: Insecure data handling
Function: `dhcpv6_parse_option_ia_na`
Component: Networking
CID: [329999](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=329999)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/dhcpv6.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-606](http://cwe.mitre.org/data/definitions/606.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

rlubos, 2023-11-17:
Similar to https://github.com/zephyrproject-rtos/zephyr/issues/65335, howerver it should be checked that `net_pkt_skip()` succeeds, otherwise unrecognized option with invalid length could exceed net_pkt size w/o noticing. Will submit a fix.

jukkar, 2023-12-22:
Re-closing as the issue is fixed.

PR #65398 (2023-11-20): net: dhcpv6: Verify net_pkt_skip() return value
Verify the return value of net_pkt_skip() function, in case the parser ignores the unrecognized options, so that in case the option was malformed and the actual provided option length exceeds the packet length, it is recognize (net_pkt_skip() should fail then).

Fixes #65347
closed 2023-12-22 by jukkar