zephyrproject-rtos/zephyr #65375: [Coverity CID: 322696] Inequality comparison against NULL in kernel/mmu.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/kernel/mmu.c

Category: Incorrect expression
Function: `z_phys_map`
Component: Kernel
CID: [322696](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=322696)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/kernel/mmu.c#L780

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_BAD_COMPARE). ([CWE-480](http://cwe.mitre.org/data/definitions/480.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

peter-mitsis, 2023-11-20:
The claim in Coverity is that 

```
Inequality comparison against NULL (BAD_COMPARE)
null_misuse: Comparing pointer dest_addr against NULL using anything besides == or != is likely to be incorrect.

```

Although I think that this specific issue is a false positive, further analysis of the code does identify three other suspect items in that code block.

1. `(dest_addr < Z_VIRT_RAM_END)` [on line 781](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L781) contains a potential off by 1 type error. As Z_VIRT_RAM_END = (Z_VIRT_RAM_START + Z_VIRT_RAM_SIZE), an unfortunate selection of values could result in that evaluating to 0.
2. There is a separate concern about `(dest_addr + aligned_size) >= Z_VIRT_RAM_START)` [on line 782](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L782) that the compiler may omit this check as it may assume that increasing the pointer will not result in a wrap-around condition and it will re-use the result from [line 780](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L780)
3. [Line 783](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L783) is a repeat of the issue on [line 781](https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/mmu.c#L781)

peter-mitsis, 2023-11-21:
Marked the Coverity issue as a false positive. The other items that I identified are distinct from this issue and will be tracked via a separate issue.

PR #65754 (2023-11-28): kernel: mmu: Fix static analysis issue
Instead of performing a set of relative address comparisons using pointers of type 'uint8_t *', we leverage the existing IN_RANGE() macro and perform the comparisons using 'uintptr_t'.

Fixes #65375
closed 2023-11-28 by nashif