zephyrproject-rtos/zephyr #65376: [Coverity CID: 322693] Explicit null dereferenced in subsys/fs/ext2/ext2_ops.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/fs/ext2/ext2_ops.c

Category: Null pointer dereferences
Function: `ext2_unlink`
Component: File System
CID: [322693](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=322693)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/fs/ext2/ext2_ops.c#L518

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

de-nordic, 2023-11-22:
> We'll take a look at this, thanks for the ping!

Sure, I will ping you on some others too.
Fill in the Coverity, if you are unable to, I will do that.

de-nordic, 2025-01-28:
@tgorochowik Please take a look at this issue.

closed 2025-01-28 by de-nordic