zephyrproject-rtos/zephyr #65377: [Coverity CID: 322680] Untrusted loop bound in subsys/fs/ext2/ext2_impl.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/fs/ext2/ext2_impl.c

Category: Insecure data handling
Function: `ext2_lookup_inode`
Component: File System
CID: [322680](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=322680)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/fs/ext2/ext2_impl.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-606](http://cwe.mitre.org/data/definitions/606.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

de-nordic, 2023-11-22:
@tgorochowik  Please take a look.

tgorochowik, 2023-12-05:
Closing as false positive (coverity updated).
All possible loop bound values could be correct, no additional sanity checks are required.

closed 2023-12-05 by tgorochowik