zephyrproject-rtos/zephyr #65378: [Coverity CID: 322661] Explicit null dereferenced in subsys/fs/ext2/ext2_ops.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/fs/ext2/ext2_ops.c

Category: Null pointer dereferences
Function: `ext2_unlink`
Component: File System
CID: [322661](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=322661)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/fs/ext2/ext2_ops.c#L518

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

de-nordic, 2023-11-22:
@tgorochowik Please take a look.

tgorochowik, 2023-11-23:
This one also seems to be a false positive.

Parent can be a null here:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7c17ac4bd0198fc9bb3a0e55aa8d5e2fae96e/subsys/fs/ext2/ext2_ops.c#L518

Only if this:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7c17ac4bd0198fc9bb3a0e55aa8d5e2fae96e/subsys/fs/ext2/ext2_ops.c#L513

returns success and parent is still a null.

It can happen only if this if is not executed:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7c17ac4bd0198fc9bb3a0e55aa8d5e2fae96e/subsys/fs/ext2/ext2_impl.c#L505-L511

but the flags we use are always set to:
https://github.com/zephyrproject-rtos/zephyr/blob/41b7c17ac4bd0198fc9bb3a0e55aa8d5e2fae96e/subsys/fs/ext2/ext2_ops.c#L511

which guarantees that parent will not be null in our case.

@de-nordic are you okay with this explanation or do you think some extra checks should be added? Some checks could be useful if the code changes in the future. This could then go unnoticed.

de-nordic, 2025-01-28:
Closed as False Positive.

closed 2025-01-28 by de-nordic