zephyrproject-rtos/zephyr #65382: [Coverity CID: 322647] Unintentional integer overflow in subsys/fs/ext2/ext2_disk_access.c

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/fs/ext2/ext2_disk_access.c

Category: Integer handling issues
Function: `disk_access_device_size`
Component: File System
CID: [322647](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=322647)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/fs/ext2/ext2_disk_access.c#L26

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

de-nordic, 2023-11-22:
@tgorochowik Please take a look.

de-nordic, 2025-01-28:
Classified as false positive and marked to be ignored.

closed 2025-01-28 by de-nordic