zephyrproject-rtos/zephyr #65384: [Coverity CID: 276746] Use of 32-bit time_t in usr/include/time.h

zephyrbot, 2023-11-16:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/usr/include/time.h

Category: High impact quality
Function: ``
Component: Other
CID: [276746](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=276746)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/master/usr/include/time.h#L76

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

henrikbrixandersen, 2023-11-17:
> Static code scan issues found in file:
> 
> https://github.com/zephyrproject-rtos/zephyr/tree/master/usr/include/time.h

Err. We don't have a "master" branch nor the file in question in the zephyr repo?

nashif, 2023-11-17:
yes, something is totally wrong here, also the issue was detected last year, so not sure why it is showing up now. could be a leak or cross contimnatin in coverity, closing.

closed 2023-11-17 by nashif