zephyrproject-rtos/zephyr #65387: ADC via DMA on STM32:  DMA device is null and `dma_config()` leads to Instruction Address Violation

doodek, 2023-11-16:
Hello Developers!

_this is my first issue to Zephyr, or any other open source project! if i should put this somewhere else or I just do it wrong -- tell me please._

I have a problem with ADC over DMA on STM32F429i-DISC1 board running samples/drivers/adc sample. I get a hardfault on dma_config(), executing an undefined instruction (api->config()). I think that something is not fully initialized through the DT.

I had the same issue on a STM32G071 on my custom board on DMAMUX and I couldn't solve it, so I tried a SoC (F429) without the MUX, but the problem prevails. My workaround was to use HAL/LL for DMA temporarily, but I would like to switch back to Zephyr implementation whenever possible. Any suggestions are much appreciated. 

I include a debug screenshot, call stack and zephyr terminal output, together with the DTS overlay and prj.conf entries.

Specs: 
Board: STM32f429i-Disc1
Build: `west build -p -b stm32f429i_disc1 samples/drivers/adc`  
Debug and flash tool: JLink OCD
Development environment: VSCode + cortex-debug
ZephyrOS version: 3.5.0
SDK: 0.16.3_windows_x86_64

### prj.conf entries:
```
CONFIG_ADC=y
CONFIG_DMA=y
CONFIG_ADC_STM32_DMA=y
CONFIG_ADC_ASYNC=y
```

### Debug screenshot (VSCode + cortex-debug via JLink)

> **Note:** `api=dev->api;` while `dev` is 0x00! Therefore `api` struct contains nonsense, and calling a function from a pointer inside it will lead to provblems

[image]

### Call Stack 
[image]

### Terminal screenshot
[image]

...and is my DTS overlay
```
 / {
    zephyr,user {
        io-channels = <&adc1 1>;//<&adc1 2 &adc1 1>;
    };
};

&adc1 {
    #address-cells = <1>;
    #size-cells = <0>;
    status = "okay";
    pinctrl-0 = <&gpio_a6_analog>;
    pinctrl-names = "default";
    st,adc-prescaler = <4>;
    st,adc-clock-source = <SYNC>;
    dmas = <&dma2 0 0 (STM32_DMA_PERIPH_TO_MEMORY | STM32_DMA_PERIPH_16BITS) 0x00>;
    dma-names="adcdma";
    channel@1 {
        reg = <1>;
        zephyr,gain = "ADC_GAIN_1";
        zephyr,reference = "ADC_REF_INTERNAL";
        zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
        zephyr,resolution = <12>;
    };

};

&pinctrl {
    gpio_a6_analog: gpio_a6_analog {
        pinmux = <STM32_PINMUX('A',6,ANALOG)>;
    };
    status = "okay";
};

&dma2 {
    status = "okay";
};
``` 

Hope putting this issue may help someone in the future

p.s. Used screenshots instead of text logs because I put the same problem on Discord forum before, and this is rather a copy of the post I made

heinwessels, 2023-11-17:
A developer who has this same issue also mentioned that they the following warning. This would mean that `CONFIG_ADC_STM32_DMA=y` as expected, but somehow the dma `struct stream` is never filled.
```
z-workspace/zephyr/drivers/adc/adc_stm32.c:713:13: warning: 'dma_callback' defined but not used [-Wunused-function]
  713 | static void dma_callback(const struct device *dev, void *user_data,
      |             ^~~~~~~~~~~~
[142/142] Linking C executable zephyr/zephyr.elf
```
This will only happen if this `DT_INST_DMAS_HAS_NAME` check fails. (Which is an odd silent failure to have. It should be better)
https://github.com/zephyrproject-rtos/zephyr/blob/bc08e801c6b1416dcbbc679f5e25f801d438a531/drivers/adc/adc_stm32.c#L1553-L1556

Which is most likely caused by the driver currently requiring a _dmamux_ to function, and not _any_ dma.
https://github.com/zephyrproject-rtos/zephyr/blob/bc08e801c6b1416dcbbc679f5e25f801d438a531/drivers/adc/adc_stm32.c#L1604-L1609

Of the top of my head I'm not sure how this should be fixed.

doodek, 2023-11-17:
That's an interesting insight! Before I have a deeper look into that, I would like to say that while this issue example is from F4 microcontroller without DMAMUX, I had the same problem on a STM32G0 with DMAMUX, used as in the documentation:
`dmas = <&dmamux1 .... >;`

heinwessels, 2023-11-17:
> That's an interesting insight! Before I have a deeper look into that, I would like to say that while this issue example is from F4 microcontroller without DMAMUX, I had the same problem on a STM32G0 with DMAMUX, used as in the documentation: `dmas = <&dmamux1 .... >;`

Digging a little deeper, I don't think that's actually the limitation. Instead the DTS **_has_** to name the dma `dmamux`, regardless of if it's an dmamux or not. Otherwise it will silently be ignored, and then you'll see the behaviour you saw which explained in my previous comment.

```dts
dma-names="dmamux";
```

Kind of a silly current limitation. Thinking of creating a PR to rename it to simply `dma`. Other drivers name it useful names like `tx` and `rx`, but these don't make sense in this case.

Also, there should be a mechanism to prevent the silent ignoring, but working that into the macro magic is not that easy. Hit my head against it now for a bit. Will see if I can maybe come up with a solution.

closed 2023-11-23 by carlescufi

Oleh-Kravchenko, 2025-07-07:
Hi @nashif, @carlescufi. 
I try to enable DMA ADC for STM32l1x, but:
```
[113/221] Building C object zephyr/drivers/adc/CMakeFiles/drivers__adc.dir/adc_stm32.c.obj
zephyr/drivers/adc/adc_stm32.c:777:13: warning: 'dma_callback' defined but not used [-Wunused-function]
  777 | static void dma_callback(const struct device *dev, void *user_data,
      |             ^~~~~~~~~~~~
```
Device tree:
```
&adc1 {
        pinctrl-0 = <&adc_in8_pb0>;
        pinctrl-names = "default";
        status = "okay";
        st,adc-prescaler = <2>;

        #address-cells = <1>;
        #size-cells = <0>;

        dmas = <&dma1 1 0 (STM32_DMA_PERIPH_RX | STM32_DMA_MEM_16BITS | STM32_DMA_PERIPH_16BITS)>;
        dma-names = "dma";

        channel@8 {
                reg = <8>;
                zephyr,gain = "ADC_GAIN_1";
                zephyr,reference = "ADC_REF_INTERNAL";
                zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
                zephyr,resolution = <12>;
        };
};
```

So broken again?
Zephyr v3.7.1

gautierg-st, 2025-07-07:
Thanks @heinwessels, L1 uses a dma-v2bis, so that's probably it.
In any case, I just tried on latest main and an ADC test with DMA compiles just fine on Nucleo L152RE.

PR #65429 (2023-11-23): driver: adc: stm32: support dma for devices without dmamux
Previously the STM32 DMA driver was dependent on a very specific
name for the DMA in the DTS. This hidden requirement has caused
a bit of confusion. This commit changes the driver to instead
always use the first DMA listed in the ADC node's dma property.

Also, updates the dma code in the ADC driver to support the STM32L4 (and likely more other mcus)

Fixes: #65387
commit f27e4547 (2023-11-23): drivers: adc: stm32: driver now agnostic of actual dma name

Previously the STM32 DMA driver was dependent on a very specific
name for the DMA in the DTS. This hidden requirement has caused
a bit of confusion. This commit changes the driver to instead
always use the first DMA listed in the ADC node's dma property.

Should fix: #65387