zephyrproject-rtos/zephyr #65391: re-consider the pytest apporach to twister

hakehuang, 2023-11-17:
**Is your enhancement proposal related to a problem? Please describe.**
we add many white box testing to twister scripts with pymock, however, this makes twsiter development work very hard, as those mock is not constructured smart enought to go with the twister code change, which makes developer have to pay a lot of time to update the pytest cases, and the time is even more than develop the feature it self.

**Describe the solution you'd like**
I would like to stop the way we adding mock testing to twisters, do not use mock to do the unit test, but use test vector vs expect test result with real twister class instanced. 

**Additional context**
take my pr example
1. https://github.com/zephyrproject-rtos/zephyr/pull/64551/files#diff-965d09e01b463ef2928c22b57c09f1412a6c12ff0bf32082c75d8ac95733f84d
2. https://github.com/zephyrproject-rtos/zephyr/pull/64551/files#diff-a0a0f8f67c7fc7229d4c2e0d9d9d91dcb351e53232d8de0f392533ed31b38ca2

the case is using mock, so any new members or reference member not init in mock, it will fail, and we need update the test code, but mock is such a big monster, it is very difficult to handle them one by one.

gchwier, 2023-11-17:
It is true that writing and understanding unit tests can take additional time, but they provide many benefits. The goal of unit testing is to save time later, by catching issues earlier and making development smoother, more reliable.

You propose tests with ‘real twister class instanced’ – but there you also need to mock / stub a lot of dependencies. It would be worth to also have such tests (as system / integration level), but do not resign from unit testing.

Maybe we should also write new tests with more ‘common sense’, not only look on the coverage (it is not the most important factor).
And it is rather not a problem of mocking. The purpose of mocking is to isolate and focus on the code being tested. When you write tests during development, it forces you to think about how your code will be used, which can result in more modular, understandable and maintainable code.
Here, in the current Twister state, it is much harder to write good tests without doing refactoring … but it is good to have such tests as a baseline. You can always decide to remove or modify some scenarios, because after refactoring (adding new features) they will not give any value.
And last point, all of us can comment, suggest something during review :)

hakehuang, 2023-11-17:
@gchwier , I am not again unit testing, my concerns are we may spend too much on maintaining the tests, take below example
```
@@ -1701,6 +1720,7 @@ def test_projectbuilder_get_binaries(
    instance_mock = mock.Mock()
    instance_mock.platform = mock.Mock()
    instance_mock.platform.binaries = platform_binaries
    +instance_mock.build_dir = ""
    env_mock = mock.Mock()

    pb = ProjectBuilder(instance_mock, env_mock, mocked_jobserver)
```
in the runner I use a variable build_dir, which is no init by mock, then we have this problem. We may run into problem that any changes in the function code, which need a class member which is OK when you go thought the normal flow, but it is not OK in mock. so we are adding extra efforts to maintain the fake scenario. We may move those mock testing in a regular base but not a must pass, and the test experts can spend time to analyze and fix the failures. We are not create a monster, that need pull every one into, just keep those in the right range.

hakehuang, 2023-11-22:
@nashif , the thing is that from my PR, I can see we introduce below issue with pymock

When using Pymock, there are potential side effects to consider:

1. **Coupling between test and test subject:** When using mocking to test a specific part, you need to have a deep understanding of the behavior of the object being mocked. This can lead to tight coupling between the testing code and the code being tested, and any changes to the mocked object may require updates to the test code.
2. **Overdependent on mocking**: Overreliance on mocking can cover up some real problems. The behavior of the mocking can sometimes differ from the actual situation, leading to potential problems being overlooked in testing.
3. **Design difficulties**: Designing and implementing mocking can take time and effort, particularly for complex behaviors or interactions with systems. 
4. **Reduced readability**: This can make it more difficult for other developers to understand testing examples, especially when the mocking and setting up of identities become complex.

hakehuang, 2023-11-23:
some more supporting 
https://www.47deg.com/blog/mocking-and-how-to-avoid-it/

nashif, 2024-01-26:
Has been discussed at lenght, do not see how this can be addressed, if there is a real issue, file a bug please.

closed 2024-01-26 by nashif