zephyrproject-rtos/zephyr #65392: Upgrade LittleFSto v2.8.0

sam131208, 2023-11-17:
**Is your enhancement proposal related to a problem? Please describe.**
The latest version of LittleFS is already V2.8.1, and after V2.5.1, the on-disk minor version of littlefs from lfs2.0 -> lfs2.1.
In some applications, you need to preload some read-only files to the file system. Commonly used tools such as littlefs-python make file system images that are not recognized by Zephyr's littlefs.

**Describe the solution you'd like**
Upgrade LittleFSto v2.8.0，or provide a tool for making file system images.

**Describe alternatives you've considered**
Provide a tool for making file system images adapted to the Zephyr version of Littlefs.

**Additional context**
*** Booting Zephyr OS build zephyr-v3.5.0-1186-g3769938e785b ***
[00:00:00.660,000] <inf> littlefs: LittleFS version 2.5, disk version 2.0
[00:00:00.661,000] <inf> littlefs: FS at flash-controller@60002000:0x240000 is 2                                                                                                                                                             112 0x1000-byte blocks with 512 cycle
[00:00:00.661,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.661,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:4155: I                                                                                                                                                             nvalid version v2.1
[00:00:00.661,000] <err> littlefs: can not format read-only system
[00:00:00.661,000] <err> fs: fs mount error (-30)
[00:00:00.661,000] <err> main: Error mounting priv littlefs [-30]

kartben, 2023-11-17:
As a temporary workaround, it looks like the FS version can be forced to 2.0 in littlefs-python: https://github.com/jrast/littlefs-python/blob/master/test/test_multiversion.py#L16 (not sure why they don't expose this through the CLI, it looks like this could be convenient).

sam131208, 2023-11-17:
Yes, I also noticed that it can be used under python, which may be convenient for users who are familiar with python.
For many people, CLI is more commonly used and a little more convenient.Unfortunately, the CLI does not provide the ability to specify the lfs version.

closed 2024-01-24 by carlescufi