zephyrproject-rtos/zephyr #65404: samples\boards\stm32\backup_sram Backup sram contents retains even after power reset for nucleo f429zi board

tanujapatgar-eaton, 2023-11-17:
After flashing samples\boards\stm32\backup_sram code for nucleo f429zi, garbage values are stored and if we reset the board stored value is increment to 1. 
But after power reset stored value is not starting from 0.
Attaching screenshot for your reference

[image]

nordicjm, 2023-11-17:
Zephyr 3.2 is not supported, please confirm this issue is still present in a supported version of zephyr, ideally `main`, then edit your post to have the required information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

erwango, 2023-11-17:
@tanujapatgar-eaton can you reproduce on recent Zephyr release ?

tanujapatgar-eaton, 2023-11-20:
@erwango reproduce on 3.4.0 same issue
[image]

FRASTM, 2023-11-27:
The fact is that the backup sram defined by the nucleo_f429 DTS is the `backup_sram: memory@40024000 `
This section in the BKPSRAM of the stm32f429 device which differs from the (RTC) backup domain .
It is mentionned in the [RefMan RM0090](https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf) : 
_The BKPSRAM is not affected by this reset. The only way of resetting the
BKPSRAM is through the Flash interface by requesting a protection level change from 1 to
0._
file:///local/home/frq09495/Pictures/Screenshots/Screenshot%20from%202023-11-27%2016-00-39.png

erwango, 2023-11-27:
> The fact is that the backup sram defined by the nucleo_f429 DTS is the `backup_sram: memory@40024000 ` This section is the **BKPSRAM** of the stm32f429 device which differs from the (RTC) backup domain . It is mentionned in the [RefMan RM0090](https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf) : _"The BKPSRAM is not affected by this reset. The only way of resetting the BKPSRAM is through the Flash interface by requesting a protection level change from 1 to 0."_
> 
> ![Screenshot from 2023-11-27 16-00-39](https://private-user-images.githubusercontent.com/51417392/285915420-216376b7-45d1-437c-b285-da08628b930a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExMDI1NDIsIm5iZiI6MTcwMTEwMjI0MiwicGF0aCI6Ii81MTQxNzM5Mi8yODU5MTU0MjAtMjE2Mzc2YjctNDVkMS00MzdjLWIyODUtZGEwODYyOGI5MzBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI3VDE2MjQwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2YjlmMjkyNmMyNjAzNGVmYTNjOWQ1MDMyZDY5MzNmMjRiYjY4ZmFiMzc2ZGJlNWU3ZTJkNjkxYmM4MjRlNWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._4lIr2gsccOpgzWbIC8kjRhaVY4QAu9sfau5phkGEkE)

ok, then is there something that should be done about it ?

FRASTM, 2023-11-28:
The stm32serie

FRASTM, 2023-11-28:
At the time the samples/boards/stm32/backup_sram/  was introduced  (by the PR #31702)
same result on stm32f429zi or stm32f746zg nucleo boards : 
```
*** Booting Zephyr OS build zephyr-v2.5.0-81-g68344cd87cba  ***
Current value in backup SRAM: 1013959382
Next reported value should be: 1013959383
Keep VBAT power source and reset the board now!
```

--> the nucleo board power off/on does not reset the content of the backup_sram

FRASTM, 2023-11-28:
@tanujapatgar-eaton What you observe is the expected hw backup_sram of the stm32f429
-    power off (on Vbat) does not reset the content of the bkup_sram (common feature of the stm32f2, stm32f7, stm32f4x)
-    reset : the content of the bkup_sram is kept like on any other stm32: stm32H5, stm32H7, stm32U5

darshanmirajkar-eaton, 2023-11-30:
> At the time the samples/boards/stm32/backup_sram/ was introduced (by the PR #31702) same result on stm32f429zi or stm32f746zg nucleo boards :
> 
> ```
> *** Booting Zephyr OS build zephyr-v2.5.0-81-g68344cd87cba  ***
> Current value in backup SRAM: 1013959382
> Next reported value should be: 1013959383
> Keep VBAT power source and reset the board now!
> ```
> 
> --> the nucleo board power off/on does not reset the content of the backup_sram If your application needs to reset the data at power on, then it should use the BOR flag as follows:
> 
> ```
> 	if (LL_RCC_IsActiveFlag_BORRST()) {
> 		backup_data = 0; 
> 		LL_RCC_ClearResetFlags();
> 	}
> ```

after making these changes in zephyr\samples\boards\stm32\backup_sram\src\main.c
[image]

power off: backup sram contents are not retaining
board reset( pressing reset button): according to the main.c application value is incrementing by 1
[image]

tanujapatgar-eaton, 2023-11-30:
> At the time the samples/boards/stm32/backup_sram/ was introduced (by the PR #31702) same result on stm32f429zi or stm32f746zg nucleo boards :
>
> ```
> *** Booting Zephyr OS build [...]

after making these changes in zephyr\samples\boards\stm32\backup_sram\src\main.c
[image]
### Observations:
power off: backup sram contents are not retaining
board reset( pressing reset button): according to the main.c application value is incrementing by 1
[image]

erwango, 2023-11-30:
Moving to enhancement as requested behavior is not the exact purpose of the sample (see README)

FRASTM, 2024-01-04:
Do you need a _new_ sample application, adapted from the existing samples/boards/stm32/backup_sram/ and where 
` "Each time the application runs the current value is displayed and then incremented by one. After power reset stored values supposed start from 0"`
Then I propose https://github.com/zephyrproject-rtos/zephyr/pull/67194

rzeeshan565, 2024-07-19:
Hi @tanujapatgar-eaton, I am trying to enable backup_sram for STM32f407VG. problem is I'm using a relatively older version of Zephyr (2.4.0). but I see you have used zephyr-v2.5.0. could you share your dts and example .c file?

After a lot of fixing dts here and there, I managed to reduce the error to this one.

error: 'DT_N_S_soc_S_memory_40024000_P_clocks_IDX_0_VAL_bus' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_timers_40014000_P_clocks_IDX_0_VAL_bus'?
 #define DT_N_INST_0_st_stm32_backup_sram DT_N_S_soc_S_memory_40024000
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~