zephyrproject-rtos/zephyr #65432: qdec in same70q21b not compiling, however in same70q21 works

tswaehn, 2023-11-18:
It looks like there is an issue with using the QDEC in **same70q21b** vs **same70q21**.

When compiling with enabled `sam-qdec` I get an issue in that file:
[drivers/sensor/qdec_sam/qdec_sam.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/qdec_sam/qdec_sam.c)

missing definition of `TC_CMR_ETRGEDG_NONE`

```
.../zephyr/drivers/sensor/qdec_sam/qdec_sam.c: In function 'qdec_sam_configure':
.../zephyr/drivers/sensor/qdec_sam/qdec_sam.c:90:27: error: 'TC_CMR_ETRGEDG_NONE' undeclared (first use in this function); did you mean 'TC_CMR_ETRGEDG_Pos'?
   90 |                         | TC_CMR_ETRGEDG_NONE

```

this definition is present in:

[asf/sam/include/same70/component/tc.h](https://github.com/zephyrproject-rtos/hal_atmel/blob/master/asf/sam/include/same70/same70q21.h)

but **not** present in:
[asf/sam/include/same70b/component/tc.h](https://github.com/zephyrproject-rtos/hal_atmel/blob/master/asf/sam/include/same70b/same70q21b.h)

steps to reproduce
cmake
`-DBOARD=sam_e70b_xplained` (**fails**)  vs `-DBOARD=sam_e70_xplained` (**works**)

dt
```
&tc0 {
	status = "okay";
	compatible = "atmel,sam-tc-qdec";

	pinctrl-0 = <&tc0_qdec_default>;
	pinctrl-names = "default";
	label = "QDEC_0";
};
```
conf
```
# quadrature decoder
CONFIG_SENSOR=y
CONFIG_QDEC_SAM=y

```
from the datasheet, I dont see any difference between both revision regarding qdec:
[image]

I suggest to copy paste from 
`same70/same70q21.h`
to
`same70b/same70q21b.h`

nandojve, 2023-11-18:
Hi @tswaehn ,

Please, open a PR on the hal and create a PR in Zephyr to update the hash.
Make sure you follow https://github.com/zephyrproject-rtos/hal_atmel#how-to-submit-code

tswaehn, 2023-11-22:
@nandojve would be great if you could have an eye on the [PR#35](https://github.com/zephyrproject-rtos/hal_atmel/pull/35)

tswaehn, 2024-01-27:
Finally I created the changes in hal_atmel
* https://github.com/zephyrproject-rtos/hal_atmel/pull/35

And also created a PR in zephyr (west.yaml) to point to the changeset in hal_atmel.

... I hope thats it?

PR #68189 (2024-01-31): west.yml: update hal_atmel to include qdec SAME70b fix
This update fixes the qdec in SAME70b, which could not be compiled prior to this fix, due to missing TC_CMR definitions. Moreover it fixes #65432.
commit 11c83e79 (2024-01-31): west.yml: update hal_atmel to include qdec SAME70b fix

This update fixes the qdec in SAME70b, which could not be compiled prior to
this fix, due to missing TC_CMR definitions. Moreover it fixes #65432.
closed 2024-01-31 by carlescufi