zephyrproject-rtos/zephyr #65438: Can we support China's self-developed CPU architecture c-sky

maweicoltsmart, 2023-11-19:
Linux already supports China's self-developed CPU architecture C-SKY. Can it be ported to Zephyr

henrikbrixandersen, 2023-11-19:
I am sure, it can - however, someone needs to put in the work. Are you interested in porting it yourself?

maweicoltsmart, 2023-11-20:
> I am sure, it can - however, someone needs to put in the work and maintain it going forward. Are you interested in porting it yourself and maintaining it for the years to come?

haha, Sorry, this job is beyond my capabilities. Does the organization have any plans to support the c-sky CPU architecture

henrikbrixandersen, 2023-11-20:
> haha, Sorry, this job is beyond my capabilities. Does the organization have any plans to support the c-sky CPU architecture

The Zephyr Project is an open source project, and as such relies on someone to step up and do the work and volunteer to maintain it. I am not aware of anybody planning to add support for C-Sky, sorry.