zephyrproject-rtos/zephyr #65446: ESP32-S3-DevKitC cannot use GPIO35~GPIO48 as a normal output.

sam131208, 2023-11-20:
**Describe the bug**
ESP32-S3-DevKitC cannot use GPIO35~GPIO48 as a normal output.
[blinky.zip](https://github.com/zephyrproject-rtos/zephyr/files/13405628/blinky.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr\samples\basic\blinky
2. add esp32s3_devkitm.overlay:
```
/ {
	aliases {
		led0 = &blue_led;
	};

	leds {
		compatible = "gpio-leds";

		blue_led: led_0 {
			gpios =  <&gpio0 35 GPIO_ACTIVE_HIGH>;
			label = "Blue - LED0";
		};
	};
};
```
3. west build -b esp32s3_devkitm
4. west flash
5. Observe the GPIO35 status.

**Expected behavior**
GPIO35~GPIO48 can output high or low normally.

**Impact**
There is no way to control the output status of the GPIO， It's always low.

**Environment (please complete the following information):**
 - OS:  Windows10
 - Toolchain ：  Zephyr SDK 0.16.3
 - The latest version of Zephyr.

**Additional context**
[blinky.zip](https://github.com/zephyrproject-rtos/zephyr/files/13405659/blinky.zip)

kartben, 2023-11-20:
Should be `&gpio1 4` in your example above I believe 
https://zephyrproject.org/how-to-set-up-esp32-gpio-pins-in-zephyr-rtos/

nordicjm, 2023-11-20:
https://github.com/zephyrproject-rtos/zephyr/discussions/65244

closed 2023-11-20 by nordicjm