zephyrproject-rtos/zephyr #65470: Native simulation IRQ handler is not always called when IRQ is raised

Fymyte, 2023-11-20:
**Describe the bug**

In simulation we are using IRQ to notify Zephyr from posix thread about some event.

We install Zephyr IRQ handler like this:
```
IRQ_CONNECT(ZN_OS_UNBLOCK_IRQ, HW_IRQ_PRIO, proxyIrqHandler, NULL, 0);
irq_enable(ZN_OS_UNBLOCK_IRQ);
```
and then call
```
hw_irq_ctrl_set_irq(ZN_OS_UNBLOCK_IRQ);
```
from posix thread to notify Zephyr. But sometimes IRQ handler is not called.

Issue seems happens because in the `boards/posix/native_posix/irq_ctrl.c`, `irq_mask`, `irq_premask` and `irq_status` variables are global and can be read and modified from multiple threads, but there is no protection on access to these variables, which leads to races.

Adding atomic protection around those variable seems to fix the issue, but it would be nice to further check which variable need those protections exactly.

**Impact**
IRQ handler not called every time.

**Environment:**
 - OS: Linux
 - Toolchain: host
 - v3.5

aescolar, 2023-11-20:
Hi @Fymyte 

> to notify Zephyr from posix thread 

I presume you mean you are spawning your own linux pthread and calling from there `hw_irq_ctrl_set_irq()`.
You are absolutely not meant to do that :)
You cannot just call into any API from any thread you may spawn.
`hw_irq_ctrl_set_irq()` is an internal API which can only be called from very specific places. That's why it does not have the kind of protection you expect. Even if it had that kind of protection, havoc would be broken in some cases down the line (it is not just about those variables not being updated atomically).

Fymyte, 2023-11-20:
Hi @aescolar

Sorry, I was not clear on this part. We are in an SMP context. POSIX threads are spawned from POSIX zephyr app, not host. We have multiple POSIX threads which interact with `hw_irq_ctrl_set_irq()` and the zephyr thread calling `hw_irq_ctrl_get_highest_prio_irq()`. This means that multiple execution flow can change `irq_status` at the same time without protection

aescolar, 2023-11-20:
What do you mean with "We are in an SMP context".
(Neither native_sim , nor the posix arch support SMP)

aescolar, 2023-11-21:
> Not SMP just multithreaded, my bad. We have multiple POSIX thread running concurrently calling `hw_irq_ctrl_set_irq()`

If you just have Zephyr threads, and are using Zephyr's native_sim target (and you haven't hacked anything in the architecture, soc, or native simulator itself), that is not a problem.

It would be much easier if you tried to provide a reproduction case, or described how your setup looks.

Your interrupt handlers may not be called because their interrupt is disabled, a higher priority one is running, the interrupt reason is cleared by some other handler..

Fymyte, 2023-11-21:
I am using the `native_posix` target, not `native_sim` and starting pthread in the zephyr application.

I will try to provide a simple case to demonstrate the behavior described, but basically we have this

```
 ---------------     ---------------
|  POSIX World  |   |    Zephyr     |
|               |   |               |
|  -----------  |   |               |
| | PThread 0 | |   |  -----------  |
| | set_irq() | |   | | Thread 0  | |
|  -----------  |   | | get_irq() | |
|               |   |  -----------  |
|  -----------  |   |               |
| | PThread 1 | |   |               |
| | set_irq() | |   |               |
|  -----------  |   |               |
 ---------------     ---------------
```

If I understood correctly, in  `native_sim`, since pthread are scheduled by zephyr (if we link with target libc), we cannot have multiple thread manipulating the `irq_status` at the same time. However, in `native_posix` the scheduler for pthread is the one of the host (Linux for me). This means that we can indeed have multiple posix thread calling `hw_irq_ctrl_set_irq()` concurrently. In this case, `irq_status` needs to be protected to avoid races.

What may happen in `native_posix`
```
PThread 0 read irq_status
PThread 1 read irq_status
PThread 0 write irq_status | (1<<0)
PThread 1 write irq_status | (1<<1) <- here we set bit 1 but not 0. The value is overriten
```

> Your interrupt handlers may not be called because their interrupt is disabled, or a higher priority one is running, or the  interrupt reason is cleared by some other handler.

In that case adding atomic protection to `irq_status` should not change the behavior.

Fymyte, 2023-11-27:
Yes we are indeed building a complex use case.
We want to build an application using POSIX APIs, so using `pthread_create`. We call `pthread_create` in the Zephyr's app main function (which will in turn create a thread on host, I agree on this). We are currently shifting to `native_sim` to avoid those issues of synchronization.

> you absolutely should not call into hw_irq_ctrl_set_irq() from that asynchronous Linux thread.

What am I supposed to do to signal zephyr from POSIX ? Currently, when a POSIX thread wants to signal that there is a job to do, it raises an IRQ, with a handler which gives a semaphore and unlocks a zephyr thread. This mechanism works great with the fix of atomically updating `irq_premask` and `irq_mask`. Otherwise, the semaphore is not always given, and the job is not always processed by zephyr.

The issue I describe is mostly what you tell me should not be done then ? We tested this use case, and it works great. But I can understand you don't want to fix this or consider it a non-issue because it is not officially supported.

aescolar, 2023-11-27:
@Fymyte 
In your description, you seem to be mixing very different scenarios, and I yet do not know what is it that you try to do.
I think you are trying to: build an embedded application which uses Zephyr's POSIX API compatibility shim, that you tried to build it in native_posix for testing (which is a) not supported, and b) prevented by kconfig). That when you tried to build it for native_posix, you hacked the zephyr kconfig so it would not prevent you from doing it. That then you ended up spawning threads on Linux instead of Zephyr. And that then you find yourself digging a deeper hole trying to notify from that thread into Zephyr's threads.

So, if you want to get some proper answers or help, I need to understand what you are trying to do, so please answer these question:

1) Are you trying to build an embedded application which uses Zephyr's POSIX API compatibility shim? 
If so,
1.A) are you trying to test that in your host with native_posix?
If so: You should switch immediately to using native_sim instead.

> What am I supposed to do to signal zephyr from POSIX ?

 2) What do you mean "from POSIX"? Do you mean from your Zephyr thread which you just created with `pthread_create()` in an embedded target or with native_sim? If so, the answer is, with any of Zephyr's primitives.
If you meant, "from the Linux thread, you created unwillingly while you were trying to create a Zephyr thread in native_posix". The answer is "You don't. You are in a wrong path."

Fymyte, 2023-11-27:
This is the minimal scenario where the behavior described is demonstrated.
In the zip files there are 2 patches (could not link them individually due to GitHub refusing `.patch` files):
1. Add a sample `samples/posix/irq` to test POSIX IRQ behavior without atomic.
2. The patch we ended up with for atomic protection around `irq_mask`, `irq_premask` and `irq_status`

[posix-irq-tests-patches.zip](https://github.com/zephyrproject-rtos/zephyr/files/13474317/posix-irq-tests-patches.zip)

I hope this can demonstrate what we have. This has been applied on top of zephyr main 2h ago without other modification.

aescolar, 2023-11-27:
@Fymyte nobody is rejecting that you can trigger what you described. What is not clear is why you would do that.
If you want any further help, please answer the questions. Otherwise please close this issue.

Fymyte, 2023-11-27:
1. Yes we are. As I said, we are currently moving to `native_sim`. It just happened that when we began to develop the application `native_posix` was the only option, thus we used that with a layer to communicate between POSIX pthread spawned in the zephyr application and zephyr kthread (when using zephyr's APIs and services was necessary). During the development we encountered the bug described and I just wanted to report it.
2. With `pthread_create()` with `native_posix`, not `native_sim` (no config prevent this, just select the target `native_posix_64`, no CMake nor compiler warning). From what you tell me this is not officially supported. I get it.

I can close this issue if this is a won't fix. Was just a bug report, no help requested. I can make a PR if you wish with the given patch. This fixed our issue without regression we could see. But I get that we have a complicated setup and this probably won't apply to anyone else.

aescolar, 2023-11-27:
Thanks @Fymyte 

> With pthread_create() with native_posix, not native_sim (no config prevent this, just select the target native_posix_64, no CMake nor compiler warning).

That is correct if you are not setting CONFIG_POSIX_API, which you would need to in an embedded target or native_sim to have `pthread_create()`.

> I can close this issue if this is a won't fix.

It is a won't-fix/not-a-bug, as using the `pthread_create()` to create Zephyr threads from the app is not supported in native_posix.
Therefore closing.

closed 2023-11-27 by aescolar