zephyrproject-rtos/zephyr #65472: ACE: ADSP: UP_SQUARED: tests/lib/c_lib/thrd/ fails.

gbarkadiusz, 2023-11-20:
**Describe the bug**
This test fails on a several platforms with different symptoms.
```
CAVS 25
tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal  ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc   ( Timeout )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.module  ( Timeout )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.notls  ( Failed )
```
```
ACE 30 PTL SIM 
tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal ( Build Failure )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.module ( Build Failure )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.notls ( Build Failure )
```
```
ACE 15 MTL 
tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.module ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.notls ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib ( Failed )
```
```
UP SQUARED 
tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal ( Failed )
tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc.notls ( Failed )
```

**To Reproduce**
The different steps for different platforms:
In general:
` twister --device-testing  --hardware-map <path_to>/map.yml -p <platform> -T tests/lib/c_lib/thrd -vv`

**Logs and console output**
`/twister-out/up_squared/tests/lib/c_lib/thrd/libraries.libc.c11_threads.minimal/handler.log`
```
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.5.0-2033-gde63da0081dd (delayed boot 500ms) ***
Running TESTSUITE libc_cnd
===================================================================
START - test_cnd_broadcast_wait
PASS - test_cnd_broadcast_wait in 0.101 seconds
===================================================================
START - test_cnd_errors
PASS - test_cnd_errors in 0.001 seconds
===================================================================
START - test_cnd_init_destroy
PASS - test_cnd_init_destroy in 0.001 seconds
===================================================================
START - test_cnd_signal_timedwait
PASS - test_cnd_signal_timedwait in 0.051 seconds
===================================================================
START - test_cnd_signal_wait
PASS - test_cnd_signal_wait in 0.051 seconds
===================================================================
START - test_cnd_timedwait_timeout
E: Timeout waiting on cond 0x128140
PASS - test_cnd_timedwait_timeout in 0.204 seconds
===================================================================
TESTSUITE libc_cnd succeeded
Running TESTSUITE libc_mtx
===================================================================
START - test_mtx_destroy
PASS - test_mtx_destroy in 0.001 seconds
===================================================================
START - test_mtx_init
PASS - test_mtx_init in 0.001 seconds
===================================================================
START - test_mtx_lock
[... 360 lines cut ...]
- FAIL - [libc_mtx.test_mtx_timedlock] duration = 0.001 seconds
- FAIL - [libc_mtx.test_mtx_trylock] duration = 0.001 seconds
- PASS - [libc_mtx.test_mtx_unlock] duration = 0.001 seconds
SUITE PASS - 100.00% [libc_once]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.001 seconds
- PASS - [libc_once.test_call_once] duration = 0.001 seconds
SUITE FAIL -  71.43% [libc_thrd]: pass = 5, fail = 2, skip = 0, total = 7 duration = 0.985 seconds
- FAIL - [libc_thrd.test_thrd_create_join] duration = 0.001 seconds
- PASS - [libc_thrd.test_thrd_current_equal] duration = 0.001 seconds
- PASS - [libc_thrd.test_thrd_detach] duration = 0.101 seconds
- FAIL - [libc_thrd.test_thrd_exit] duration = 0.001 seconds
- PASS - [libc_thrd.test_thrd_reuse] duration = 0.003 seconds
- PASS - [libc_thrd.test_thrd_sleep] duration = 0.877 seconds
- PASS - [libc_thrd.test_thrd_yield] duration = 0.001 seconds
SUITE PASS - 100.00% [libc_tss]: pass = 2, fail = 0, skip = 0, total = 2 duration = 0.002 seconds
- PASS - [libc_tss.test_tss_create_delete] duration = 0.001 seconds
- PASS - [libc_tss.test_tss_get_set] duration = 0.001 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: ae16559e6dc6c32cd229b2b0caa26312
PROJECT EXECUTION FAILED
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.3
 - Commit SHA zephyr-v3.5.0-2033-gde63da0081dd

nashif, 2023-11-21:
This is suspected to be SMP related, i.e. it passes on some of those platforms with CONFIG_SMP=n

nashif, 2023-11-21:
@peter-mitsis  FYI

peter-mitsis, 2023-11-22:
FWIW, I have reproduced the failures on the intel_adsp_ace15_mtpm. Those, failures _could_ be SMP related, however the output is raising an eyebrow ...

```
E: Timeout locking mutex 0xnnnnnnnn  # Passing case
vs
E: Mutex is invalid (nnnnnnnn)           # Failing case
```

Why would we be getting an invalid mutex message? I could be wrong, but my spidey-sense suspects that an incoherent cache is coming into play--that is, one CPU is attempting to access the stack of a thread running on another CPU. (That problem does not exist on uniprocessor xtensa systems).

Digging into this ....

peter-mitsis, 2023-11-22:
The xtensa cache thing is coming into play again. Take test_mtx_timedlock() for example. It creates a mutex on the stack of one thread, and then accesses it from another. Since on an Xtensa SMP system, that is likely to be on another CPU we have the not-so-good scenario of a thread on one CPU accessing stack memory of a thread running on another CPU (not so good because that memory is guaranteed to be kept in sync in that circumstance).

To get around this, either we would need to fix the tests so that the objects being tested are global objects (and thus not on the stack), or prevent this set of tests from being executed on xtensa platforms (which would arguably leave a testing hole).

As for the up_squared failure ... I have not delved into that one.

cfriedt, 2023-11-23:
> To get around this, either we would need to fix the tests so that the objects being tested are global objects (and thus not on the stack), or

That's exactly what I was thinking. I kind of suspected it was something like that. Not an extensive xtensa user unfortunately.

@peter-mitsis - Do you want to go for it, or should I?

closed 2023-11-28 by peter-mitsis