zephyrproject-rtos/zephyr #65489: [RISC-V] drivers: intc_plic: Hardcodes enables interrupts for context 0 only, cannot enable interrupts on cores.

con-pax, 2023-11-20:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
Changes introduced in commits 8db1a5a, 0e6bef2 and 39433f06 hard code the values for irq enable offset for context 0 in the PLIC. This is a show stopper for multicore platforms that want to run a zephyr application on a single core, other than on the first core. A wider problem is that it only considers a single context to handle interrupts, but what about SMP applications?
 As an example, Polarfire SoC  utilizes the different contexts for the PLIC for the different modes (Machine, Supervisor ) such that
```
context 0 = hart 0 machine mode interrupt enables, 
context 1 = hart 1 machine mode interrupt enables,
context 2 = hart 2 supervisor mode interrupt enables
.
.
.
context x = hart 4 supervisor mode interrupt enables
 ```

A previous solution for a single core application to run on, for instance a HART other than HART 0 was to use a devicetree overlay which describes the PLIC's registers. an example (for HART 1)
```
&plic {
	reg = <0x0c000000 0x00002000
			0x0c002080 0x001fe000
			0x0C201000 0x3e000000>;
};

```
The above would enable interrupts for HART 1 (PLIC context 1). However, this solution is now broken.

**To Reproduce**
For a multi core platform such as mpfs_icicle, compile any sample application that uses PLIC interrupts. in prj.conf, set CONFIG_RV_BOOT_HART=1 (or 2, 3, etc)
The PLIC will not service the interrupt

**Expected behavior**
Any given core in a multi core platform should have its interrupts enabled.

**Impact**
Show stopper

**Environment (please complete the following information):**
 - OS: Linux, 
 - Toolchain Zephyr SDK,  0.16.3
 - ad788ad51173c08dc237653df132d500b0e40d26

cfriedt, 2023-11-21:
cc @ycsin

cfriedt, 2023-11-21:
Just did a brief survey about how other architectures / SoCs handle this today in Zephyr and it's very much ad-hoc. There is no conventional way to signal this in Zephyr. Almost every architecture and SoC can do this quite differently.

The solution that previously worked also seems to fall into that category. From what others have said, Zephyr has only technically been able to run a separate application on another core from Hart 0, so it seems that Zephyr was effectively tricked into thinking that the Hart 1 was Hart 0 via DT.

It would be very good to get some guidance from RISC-V arch maintainers here (and possibly also to explicitly add some documentation about this scenario).

Personally, I would very much like to see support for the PolarFire SoC land upstream. Thanks @con-pax for doing the legwork, and also looking forward to how this gets resolved.

ycsin, 2023-11-22:
I think this is more like an enhancement than a bug, the previous and current PLIC driver never had the notion of context/hart, it's just that the previous dts binding allowed the driver to be 'hacked' and use the addr of hart 1 as hart 0, and it doesn't do that anymore as we move the implementation into the driver itself

con-pax, 2023-11-22:
Hi @cfriedt , @ycsin 
Thank you for taking the time to reply. I'm really happy that there are eyes on this now and hopefully we can resolve it soon. I'm certainly looking at it, from our end. 

@ycsin respectfully, I have to disagree: this is a regression. I would like to explain by highlighting some points.
1. there should be no correlation between Zephyr's CPU's and HART ID's. 
On a heterogenous core complex, the HART's assigned to Zephyr may not be zero indexed and may not even be contiguous.  
2. Zephyr may not be the only OS or application running on a multi core complex. We need a way to choose the correct context (in terms of the HART and the privileged mode of that HART) that should enable interrupts for the Zephyr application and not interfere with the rest of the core complex. 
3. Previously, the "reg-names" property list for the plic included "irq_en". To me, its perfectly valid and not a hack to use an overlay to modify that given property for a specific HART's IRQ_EN, given that it was implemented in that way. 

One last point. I don't think the binding for the PLIC is fully suitable. I believe it would be benefit greatly with having a way to list the contexts used for a given platform, given that they should be platform specific.

I see that you have created a PR for a workaround. I will test this today and get back to you.

Again, I am really happy that there are people looking into this and I definitely feel that we can get it into better shape than we found it.

Kind Regards,
Conor

con-pax, 2023-11-22:
corss-posting the comment I left on PR #65553 so everyone is in the loop:

> Hi @ycsin,
> 
> Thanks for taking the time to try this. I tested this out locally (in our case: PolarFire SoC,) and the outcome is it will work when CONFIG_RV_BOOT_HART=1 (u54_1). However, It does not work when CONFIG_RV_BOOT_HART=4 (u54_4). The reason is because the privilege modes contexts are not taken into consideration. you can see in the [SiFive FU540-C000 Manual](https://static.dev.sifive.com/FU540-C000-v1.0.pdf) and [PolarFire SoC TRM](https://ww1.microchip.com/downloads/aemDocuments/documents/FPGA/ProductDocuments/ReferenceManuals/PolarFire_SoC_FPGA_MSS_Technical_Reference_Manual_VC.pdf) in the PLIC chapter:
> 
> ```
> Context 0 is for HART 0 M Mode Enables,
> Context 1 and 2 are for HART 1 M and S Mode Enables,
> Context 3 and 4 are for HART 2 M and S Mode Enables, 
> Context 5 and 6 are for HART 3 M and S Mode Enables,
> Context 7 and 8 are for HART 4 M and S Mode Enables,
> ```
> 
> This solution is a step in the right direction, however the best solution should take into consideration SMP, where each CPU/HART can handle interrupts.
> 
> Kind Regards, Conor

con-pax, 2023-11-24:
Another cross-post:
> Hey @ycsin,
> 
> Apologies for the delay. I was sidetracked yesterday. I tested the change and unfortunately it produces a system failure during the plic_init function.
> 
> I have been working on a solution today. Its definitely a "Work in Progress" but I thought I would share it with you now: https://github.com/polarfire-soc/zephyr/tree/mpfs-icicle_plic_multi-context_support

So to keep this issue up-to-date: I have been working on a solution which is working albeit, not finished yet. I misunderstood the binding for the PLIC. Essentially, the PLIC's interrupts-extended property denotes the parents (i.e, cpu 0 - 4 on PolarFire SoC). However, for PolarFire SoC, the Devicetree is not fully described (yet). 

https://github.com/polarfire-soc/zephyr/tree/mpfs-icicle_plic_multi-context_support

The above branch is the beginning of an implementation to address this issue.
From here, I include all 9 contexts in the interrupts-extended. I can then get the total number of contexts and enable the local irq for each context in an SMP cluster. I have not finished the work yet. but this solution by and large seems to work (and is similar to what is done for Linux)

con-pax, 2023-11-29:
Hi @ycsin, @cfriedt,

I have opened a PR #65922, which implements a fix. 
Cheers!

con-pax, 2023-12-12:
I am happy to close this issue now. Thanks to everyone involved for the help and support!

closed 2023-12-12 by con-pax