zephyrproject-rtos/zephyr #65504: Two SPI device control errors with one cs-gpio pin.

2songh, 2023-11-21:
**To Reproduce**

Zephyr Version: 3.4
Board: STM32L433
SPI Device: MAX31865 (1, 2)

Steps to reproduce the behavior:
1. west build -b "stm32l433"
2. west flash

**Describe the bug**

I'm trying to use two spi devices in reverse mode using one chip select pin.

* device tree
[image]

When configured like this and spi_read() on device 1 and device 2,
Action is performed only with CS_ACTIVE_LOW.

[image]

This causes problems with the return values ​​of the two devices being the same.
I want CS_ACTIVE_LOW and CS_ACTIVE_HIGH to alternate.

There is no problem when configuring and using the devices one by one.

* **CS_ACTIVE_LOW**
[image]
[image]

* **CS_ACTIVE_HIGH**
[image]
[image]

Please give me a solution or advice on this issue.

fabiobaltieri, 2023-11-21:
That's a bit of an odd use case, I think  you may make it work by writing a driver implementing the GPIO API handling the two chip select as if they were two different GPIOs and then driving a single GPIO as an output with whatever logic you need. Prob not worth upstreaming though, it's a bit of an odd use case, not sure we want to encourage that.

fabiobaltieri, 2023-11-21:
Would not call this a bug though.

2songh, 2023-11-22:
> That's a bit of an odd use case, I think you may make it work by writing a driver implementing the GPIO API handling the two chip select as if they were two different GPIOs and then driving a single [...]

First of all, thank you very much for answering.
I have a few additional questions.

We already used this method in Zephyr version 2.4, it was no problem.
Has anything changed since the version was upgraded?

*****

It shares our circuit using two MAX31865 sensors.

[image]

*****

These two share the same SPI Line.

[image]

*****

**Question**
1. Are there any differences between versions 2.4 and 3.4?
2. Is there any way to control it using one CS gpio?
3. Is adding a new CS Pin the best option?

fabiobaltieri, 2023-11-22:
>     1. Are there any differences between versions 2.4 and 3.4?

Yeah one or two things:

```console
$ git diff --shortstat v2.4.0..v3.4.0
 27599 files changed, 2294837 insertions(+), 671014 deletions(-)
```

>     2. Is there any way to control it using one CS gpio?

If it used to work you could try and bisect, see if you can find what broke it, maybe something to do with `SPI_STM32_USE_HW_SS`?

>     3. Is adding a new CS Pin the best option?

If you have pins available it's certainly the more conventional option and less likely to cause issues, plus it's usually not a great idea to keep an SPI device selected all the time.

tbursztyka, 2023-11-27:
Sounds like an STM32 issue to me (pehaps a gpio one, though unlikely)

2songh, 2023-11-28:
> Sounds like an STM32 issue to me (pehaps a gpio one, though unlikely)

Do you have any experience testing with boards other than STM32?

FRASTM, 2024-02-06:
@2songh can you please check when the DTS has no spi2_nss declared in the <pinctrl-0> but <cs-gpios>
--> set either the CS by spi pin NSS _or_ GPIO control (not both)

closed 2024-04-22 by github-actions[bot]