zephyrproject-rtos/zephyr #65511: CI: tests/subsys/logging/log_backend_uart tests fail on qemu_cortex_m0

henrikbrixandersen, 2023-11-21:
**Describe the bug**
The following tests fail on `qemu_cortex_m0´ in CI on `main`:
- tests/subsys/logging/log_backend_uart/logging.backend.uart.multi on qemu_cortex_m0 failed (Timeout)
- tests/subsys/logging/log_backend_uart/logging.backend.uart.single on qemu_cortex_m0 failed (Timeout)
- tests/subsys/logging/log_output/logging.output.thread_id on qemu_cortex_m0 failed (Timeout)

**To Reproduce**
Steps to reproduce the behavior:
1. `west twister -p qemu_cortex_m0 -s tests/subsys/logging/log_backend_uart/logging.backend.uart.multi`
2. See error as on https://github.com/zephyrproject-rtos/zephyr/actions/runs/6933236823/job/18859011016

**Expected behavior**
CI tests pass.

**Impact**
Multiple CI failures on `main`.

**Environment (please complete the following information):**
 - Commit SHA: 934a09550dd17406426ac92aac89761cbe506520

ycsin, 2023-11-21:
I tried one of the test and it was passing in bc08e801c6b1416dcbbc679f5e25f801d438a531, but fails in a43e516ff2f7bc7091835bf9a97c979937a75366

nordic-krch, 2023-11-21:
There is a similar failure reported #65456 for which I opened #65526. But cortex-m0 failures seems to related to https://github.com/zephyrproject-rtos/zephyr/commit/7a5fcb8c60e2e6ba26704fabc28fe2d4e0106513

keith-packard, 2023-11-21:
Could this be caused by CI using SDK version 0.16.3?

keith-packard, 2023-11-21:
Yup, SDK 0.16.3 has the version of picolibc without separate long long support, so making that work has broken several test cases. Fixed in #65539 (which I've added another patch to).

PR #65539 (2023-11-21): tests/kernel/mem_protect/mem_map: Reduce printf size for qemu_x86_tiny
When using picolibc before 1.8.5, the only way to get 'long long' support was to use the full version, including floating point support. This is too large for this testcase.

Reduce the size of the printf code by switching to the version without 64-bit integer support. This allows the tests to pass when using older picolibc versions, such as that included with SDK version 0.16.3.

Closes: #65509
Closes: #65511
closed 2023-11-21 by henrikbrixandersen