zephyrproject-rtos/zephyr #65512: spi_transceive weird behaviour

krupis, 2023-11-21:
I am working on zephyr RTOS on the Thingy 91 device. I am trying to read ADXL362 MEMS device ID.

According to the ADXL362 datasheet, the register should be read as shown below:
[image]

As described above, I need to send 2 bytes (instruction + address) and then read out the data
[image]

My main.c code:
```
#include <stdio.h>
#include <stdlib.h>
#include <zephyr/kernel.h>
#include <zephyr/drivers/spi.h>
#include <zephyr/device.h>
#include <zephyr/drivers/gpio.h>

#define LOG_LEVEL CONFIG_LOG_DEFAULT_LEVEL
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(main);

#define ADXL362_WRITE_REG 0x0A
#define ADXL362_READ_REG 0x0B
#define ADXL362_READ_FIFO 0x0D

#define ADXL362_REG_DEVID_AD 0x00
#define ADXL362_REG_DEVID_MST 0x01

static int readRegister(uint8_t reg, uint8_t *values, uint8_t size);

/* Devicetree node corresponding to the peripheral to be used via Zephyr SPI
 * driver (SPIM1), in the background transfer.
 */
#define SPI_DEV_NODE DT_NODELABEL(spi3)
const struct device *const spi_dev = DEVICE_DT_GET(SPI_DEV_NODE);

static const struct spi_config spi_dev_cfg = {
	.operation = SPI_OP_MODE_MASTER | SPI_WORD_SET(8) |
				 SPI_TRANSFER_MSB,
	.frequency = 1000000,
	.cs = {
		.gpio = GPIO_DT_SPEC_GET(SPI_DEV_NODE, cs_gpios),
	},
};

int main(void)
{
	int err;
	printk("Program started \n");

	if (!device_is_ready(spi_dev))
	{
		printk("%s is not ready\n", spi_dev->name);
		return 0;
	}

	uint8_t values[1];
	while (1)
	{
		int ret = readRegister(ADXL362_REG_DEVID_MST, values, 1);
		if (ret == 0)
		{
			printk("Read chip ID failed \n");
			k_msleep(1000);
		}
		else
		{
			printk("Register chip ID: %d\n", values[0]);
			k_msleep(1000);
		}
	}
	return 0;
}

// According to the ADXL362 datasheet (https://www.analog.com/media/en/technical-documentation/data-sheets/adxl362.pdf)
// Figure 36, we follow multi byte structure where first byte is ADXL362_READ_REG and second byte is the register address
static int readRegister(uint8_t reg, uint8_t *values, uint8_t size)
{
	int err;
	uint8_t tx_buffer[2];
	tx_buffer[0] = ADXL362_READ_REG;
	tx_buffer[1] = reg;

	struct spi_buf tx_spi_buf = {
		.buf = tx_buffer,
		.len = 2};

	struct spi_buf_set spi_tx_buffer_set = {
		.buffers = &tx_spi_buf,
		.count = 1};

	struct spi_buf rx_spi_buf = {
		.buf = values,
		.len = size};

	struct spi_buf_set spi_rx_buffer_set = {
		.buffers = &rx_spi_buf,
		.count = 1};

	err = spi_transceive(spi_dev, &spi_dev_cfg, &spi_tx_buffer_set, &spi_rx_buffer_set);
	if (err)
	{
		printk("SPI error: %d\n", err);
		return 0;
	}
	return 1;
}
```

Logic analyzer : 
[image]

As you can see from my logic analyzer signal, there is some issue with the MOSI. I do not fully understand why is there gap between the first byte (0x0B and the second byte 0x01) ? Why is it not a continuous stream? Appreciate any tips, I have been stuck on this for a while.

anangl, 2023-11-21:
You need to use two RX buffers:
```
	struct spi_buf rx_spi_buf[] = {
		{
			.buf = NULL,
			.len = sizeof(tx_buffer)
		},
		{
			.buf = values,
			.len = size
		}
	};

	struct spi_buf_set spi_rx_buffer_set = {
		.buffers = rx_spi_buf,
		.count = ARRAY_SIZE(rx_spi_buf)
	};
```
But why not use the [ADXL362 sensor driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/adxl362)?

krupis, 2023-11-21:
@anangl 
Appreciate your response but please can you elaborate more? Why would I need to use 2 RX buffers for receiving one byte? Why would my current solution would not be suficient?

How come nothing is mentioned here:
https://docs.zephyrproject.org/apidoc/latest/structspi__buf.html

Regarding the sensors driver for the ADXL362, I got frustrated trying to setup trigger on motion to the point where I decided to create my own driver for it. Perhaps i will create a new issue about this but lets not dive too much here about it. Lets discuss more about using zephyrs spi calls as its not clear at all for me :(

krupis, 2023-11-27:
> For each byte sent, there would be a byte received. But obviously in the protocol you have to follow, you don't care about the 2 first bytes received (line is set to plain flat, but the controller will assume something is coming, most likely 0's but it could be garbage as well I don't know this device). Anyway, thus why you need 2 RX buffers The first RX buffer is basically like /dev/null for the first 2 bytes, and the 2nd buffer if for the value you care about.
> 
> Look at existing SPI device drivers (perhaps something like samples/drivers/spi_fujitsu_fram or actual sensor drivers such as drivers/sensor/bme280/ )
> 
> The wave form is not going to be exactly as the datasheet, depending on how the SPI controller and its driver work. The timings depend on how fast the SPI controller is fed with bytes to send (interruption, handling, feeding the controller buffer(s) etc...). That's why SPI CLK line is for: when it is not running, the device at the other end is not going to do anything either.

Thanks for explaining that. I have written a couple of SPI drivers for ESP32 and STM32 and never had to deal with any of that. Its weird Zephyr chose this approach but it is clear now.

Thanks!

harristomyx, 2023-11-29:
SPI is a full duplex protocol, you can receive and send at the same time. In your case, you require the master to send 2 bytes, during which period there are no bytes to be received, and then you receive the data out, during which nothing is sent.